intel/ros_openvino_toolkit #5: catkin_make error object_msgs

oeyyey, 2019-01-07:
Ubuntu 16.04 , ROS Kinetic
```
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "object_msgs" with
  any of the following names:

    object_msgsConfig.cmake
    object_msgs-config.cmake

  Add the installation prefix of "object_msgs" to CMAKE_PREFIX_PATH or set
  "object_msgs_DIR" to a directory containing one of the above files.  If
  "object_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  ros_openvino_toolkit/dynamic_vino_lib/CMakeLists.txt:25 (find_package)

```

chaoli2, 2019-01-08:
@oeyyey 
Have you cloned the [object_msgs ](https://github.com/intel/object_msgs) project into _~/catkin_ws/src_? The error message seems that no object_msgs is found.

oeyyey, 2019-01-08:
@chaoli2 
Have another problem 
`/catkin_ws/src/ros_openvino_toolkit/dynamic_vino_lib/include/dynamic_vino_lib/factory.h:33:43: fatal error: mkldnn/mkldnn_extension_ptr.hpp: No such file or directory
`
I have already installed mkdnn (https://github.com/intel/mkl-dnn). No idea what I missed.

chaoli2, 2019-01-08:
@oeyyey 
mkldnn headers is included in the OpenVINO toolkit. For example, if you install OpenVINO with tarball following this [guide](https://github.com/intel/ros_openvino_toolkit/blob/master/doc/BINARY_VERSION_README.md), mkldnn_extension_ptr.hpp is in /opt/intel/computer_vision_sdk/deployment_tools/inference_engine/include/mkldnn/mkldnn_extension_ptr.hpp.

So make sure your OpenVINO env is setup correctly with:
```source /opt/intel/computer_vision_sdk/bin/setupvars.sh```
this header could be found by the compiler. Will you please have a try and give me feedback whether it works?

oeyyey, 2019-01-08:
@chaoli2 
I tried to source setupvars.sh it's still not working.

chaoli2, 2019-01-09:
@oeyyey 
Did you install OpenVINO with tarball or source code? Could you find the "mkldnn_extension_ptr.hpp" under your OpenVINO installation directory?

chaoli2, 2019-01-16:
@oeyyey 
How about your environments? Does it works now?

oeyyey, 2019-01-16:
@chaoli2 
Yes, I found mkldnn_extension_ptr.hpp  and after restart ubuntu catkin_make found too but dynamic_vino_lib still have a problem with link target so I designed to make my own version of ROS openvino and move to R5. Anyway thank you for your help :)

closed 2019-01-16 by oeyyey