intel/ros_openvino_toolkit #6: Running YOLOV2/V3 on UHD 630 OpenVino, how to?

fiorano10, 2019-01-25:
After generating pb files from yolov3, how can I use it to run with ROS? Running Ubuntu 16.04, i7-8700T

chaoli2, 2019-01-28:
@fiorano10 
OpenVINO toolkit provides the Model Optimizer to convert pb files to OpenVINO IR. Here is the script for your reference:
```
 python3 /opt/intel/computer_vision_sdk/deployment_tools/model_optimizer/mo_tf.py --input_model _xxx_.pb  --tensorflow_use_custom_operations_config /opt/intel/computer_vision_sdk/deployment_tools/model_optimizer/extensions/front/tf/yolo_v3.json --tensorflow_object_detection_api_pipeline_config pipeline.config --reverse_input_channels --output_dir ./output/
```
Also attach the complete [tutorial](https://software.intel.com/en-us/articles/OpenVINO-ModelOptimizer).

fiorano10, 2019-01-28:
I have the samples working, and have XML files for yolo_v2 and yolo_v3 generated. After I update the model I get this error:
`[ ERROR ] Error reading network: cannot parse future versions: 4`
`/teamcity/work/scoring_engine_build/releases_openvino-2018-r3/src/inference_engine/ie_cnn_net_reader_impl.cpp:147
/opt/intel/computer_vision_sdk_2018.3.343/deployment_tools/inference_engine/include/details/ie_exception_conversion.hpp:80`
`[pipeline_with_params-2] process has died [pid 27048, exit code 1, cmd /home/burro/catkin_ws/devel/lib/dynamic_vino_sample/pipeline_with_params __name:=pipeline_with_params __log:=/home/burro/.ros/log/8e736be4-234a-11e9-9757-00045f7224ad/pipeline_with_params-2.log].`
`log file: /home/burro/.ros/log/8e736be4-234a-11e9-9757-00045f7224ad/pipeline_with_params-2*.log`

Also, how can I run object detection using YOLO and it's classes using OpenVino?

chaoli2, 2019-01-29:
@fiorano10 
It seems like this issue is caused by the OpenVINO version. Currently, this project only supports OpenVINO R3. Have you ever using OpenVINO R4 in your system before? If so, will you clean R4 and try it again?

fiorano10, 2019-01-29:
I did a fresh install of Ubuntu, OpenVino R3, and ROS. I still have the same issue. Here are the generated IR models for YOLOV2/3
[yolo_v2](https://drive.google.com/drive/folders/1Ze5Hzpz0cwcrxfEKl0ujssPyLxMnlOq2?usp=sharing) 
[yolo_v3](https://drive.google.com/drive/folders/121Sq_jFii6PVVX9nSsJ53EZwP4GhjhxA?usp=sharing)

chaoli2, 2019-01-30:
@fiorano10 
According to your yolov2 and yolov3 IR, the problem is caused by the version "4".
[image]
I checked the [source code](https://github.com/opencv/dldt/blob/eae43f84291492e5e6094eb7efa6077f68d7aca8/inference-engine/src/inference_engine/ie_cnn_net_reader_impl.cpp#L136) of OpenVINO inference engine and find that it will throw exception when version="4". Will you please manually edit the version to "2" or "3" to have another try? Thanks.

fiorano10, 2019-01-30:
When I change the version to '2', I get the following error:
`[ ERROR ] Error reading network: No such parameter name 'kernel-x' for layer 0-convolutional`
`/opt/intel/computer_vision_sdk_2018.3.343/deployment_tools/inference_engine/include/ie_layers.h:425`
`/opt/intel/computer_vision_sdk_2018.3.343/deployment_tools/inference_engine/include/details/ie_exception_conversion.hpp:80`

From the source code, it can't run versions greater than 2 so when I change it to '3', I get the same error as earlier:
`[ ERROR ] Error reading network: cannot parse future versions: 3`
`/teamcity/work/scoring_engine_build/releases_openvino-2018-r3/src/inference_engine/ie_cnn_net_reader_impl.cpp:147`
`/opt/intel/computer_vision_sdk_2018.3.343/deployment_tools/inference_engine/include/details/ie_exception_conversion.hpp:80`

fiorano10, 2019-01-30:
I generated a new model with the voc dataset with 20 classes but I get the following errors [yolo_voc](https://drive.google.com/drive/folders/1uBrwyQn5bCCrYxH4conW7bwFAp2NC0nd?usp=sharing):

`[ INFO ] Reading input`
`InputDvice: RealSenseCamera`
`[ INFO ] Loading network files`
`[ INFO ] Batch size is set to  1`
`[ INFO ] Checking input size`
`[ INFO ] Checking output size`
`[ INFO ] Checking Object Detection outputs`
`[ INFO ] Checking Object Detection outputs ...`
`[ INFO ] Checking Object Detection output ... Name=output/YoloRegion`
`[ INFO ] Checking Object Detection num_classes`
`[ ERROR ] Object Detection network output layer (output/YoloRegion) should have num_classes integer attribute`

When I change the classes atr to num_classes I get the following errors:

`[ INFO ] Checking Object Detection outputs ...`
`[ INFO ] Checking Object Detection output ... Name=output/YoloRegion`
`[ INFO ] Checking Object Detection num_classes`
`[ INFO ] Checking Object Detection output ... num_classes=20`
`[ INFO ] max proposal count is: 110`
`[ ERROR ] Object Detection network output layer should have 7 as a last dimension`

fiorano10, 2019-01-30:
I generated another model with 80 classes and tested it on images from [this ](https://github.com/feng1sun/YOLO-OpenVINO) and it worked. [Here](https://drive.google.com/drive/folders/1FkkF1JDsgjDC7LTmWzE8e2zWCUBqr_ZV?usp=sharing) is the new model and the output.

I get the same errors as mentioned above, first for attr num_classes and then for the network layer last dimension.

fiorano10, 2019-01-30:
After hack-fixing object_detection.cpp I get this error:

`InputDvice: RealSenseCamera`
`[ INFO ] Loading network files`
`[ INFO ] Batch size is set to  1`
`[ INFO ] Checking input size`
`[ INFO ] Checking output size`
`[ INFO ] Checking Object Detection outputs`
`[ INFO ] Checking Object Detection outputs ...`
`[ INFO ] Checking Object Detection output ... Name=output/YoloRegion`
`[ INFO ] Checking Object Detection num_classes`
`[ INFO ] Checking Object Detection output ... num_classes=80`
`[ INFO ] max proposal count is: 110`
**`[ ERROR ] No such parameter name 'classes' for layer output/YoloRegion`
`/opt/intel/computer_vision_sdk/deployment_tools/inference_engine/include/ie_layers.h:425`
`/teamcity/work/scoring_engine_build/releases_openvino-2018-r3/src/mkldnn_plugin/nodes/mkldnn_generic_node`
`/opt/intel/computer_vision_sdk_2018.3.343/deployment_tools/inference_engine/include/details/ie_exception_conversion.hpp:80`**

chaoli2, 2019-01-31:
Will you please provide the download links of all the models that you want to convert? I'd like to try to generate and test them from my side.

fiorano10, 2019-01-31:
[Here](https://drive.google.com/drive/folders/1n5xjVWplQn85yj81j2aZUTIb65yQyEGA?usp=sharing) is the link for YoloV2 weights and cfg for voc and coco.

chaoli2, 2019-03-22:
Close this thread since yolo support issue will be tracked [here](https://github.com/intel/ros_openvino_toolkit/issues/10).

closed 2019-03-22 by chaoli2