intel/ros_openvino_toolkit #7: Object detection doesn't run on GPU

fiorano10, 2019-01-28:
I'm trying to run object detection sample on the GPU but it defaults to using the CPU, why?
'[ INFO ] --------parameters DUMP---------------------
[ INFO ] Pipeline: object
[ INFO ] 	Inputs: RealSenseCamera, 
[ INFO ] 	Outputs: ImageWindow, RosTopic, RViz, 
[ INFO ] 	Inferences: 
[ INFO ] 		Name: ObjectDetection
[ INFO ] 		Model: /opt/intel/computer_vision_sdk/deployment_tools/intel_models/person-vehicle-bike-detection-crossroad-0078/FP32/person-vehicle-bike-detection-crossroad-0078.xml
[ INFO ] 		Engine: **GPU**
[ INFO ] 		Label: to/be/set/xxx.labels
[ INFO ] 	Connections: 
[ INFO ] 		ObjectDetection->ImageWindow
[ INFO ] 		ObjectDetection->RosTopic
[ INFO ] 		RealSenseCamera->ObjectDetection
[ INFO ] Common:
[ INFO ] 	camera_topic: 
[ INFO ] 	custom_cpu_library: 
[ INFO ] 	custom_cldnn_library: 
[ INFO ] 	enable_performance_count: 0
getting instance
return instance
[ INFO ] device_FACE:**CPU**
[ INFO ] model_FACE:/opt/intel/computer_vision_sdk/deployment_tools/intel_models/person-vehicle-bike-detection-crossroad-0078/FP32/person-vehicle-bike-detection-crossroad-0078.xml
[ INFO ] device_AG:**CPU**
[ INFO ] model_AG:
[ INFO ] model_HeadPose:
[ INFO ] device_HeadPose:**CPU**

chaoli2, 2019-01-29:
@fiorano10 
The default [yaml ](https://github.com/intel/ros_openvino_toolkit/blob/master/vino_launch/param/pipeline_people_tarball.yaml)for people detection combines 4 inference instances for face, emotion, age & gender and head pose respectively.  Each inference will involve one inference engine like CPU or GPU. 

If you'd like to do object detection, just keep one inference instance and remove others. Here is one object detection [yaml ](https://github.com/intel/ros2_openvino_toolkit/blob/master/sample/param/pipeline_object.yaml)example for your reference.

fiorano10, 2019-01-29:
I already have the object detection running on CPU, but even if I switch it to GPU in the yaml, the inference still defaults to CPU as I mentioned above. When I run the object detection the CPU usage jumps to 600%.

chaoli2, 2019-01-30:
@fiorano10 
Will you show me your yaml file? 
According to your provided log, the inference engine launched 4 inferences at the same time. One is running object detection on GPU, and another three inferences are running detections on CPU, such as face, age&gender and headpose detection.

fiorano10, 2019-01-30:
This is what yaml file looks like:
`Pipelines:
- name: object
  inputs: [RealSenseCamera]
  infers:
    - name: ObjectDetection
      model: /opt/intel/computer_vision_sdk/deployment_tools/intel_models/person-vehicle-bike-detection-crossroad-0078/FP32/person-vehicle-bike-detection-crossroad-0078.xml
      engine: CPU
      label: to/be/set/xxx.labels
      batch: 16
  outputs: [ImageWindow, RosTopic, RViz]
  confidence_threshold: 0.2
  connects:
    - left: RealSenseCamera
      right: [ObjectDetection]
    - left: ObjectDetection
      right: [ImageWindow]
    - left: ObjectDetection
      right: [RosTopic]

OpenvinoCommon:`

chaoli2, 2019-01-30:
@fiorano10 
The yaml is OK.
But I have not met this issue from my side. The runtime inference engine is exactly what I specified in the yaml... 
The provided log is the output of above yaml?

fiorano10, 2019-01-30:
Yup, that's the output of the above yaml file. when I run face or emotion detection, it runs fine on the GPU but not object detection

fiorano10, 2019-01-31:
You need add `FLAGS_d = pipelines[0].infers[0].engine;` in `/ros_openvino_toolkit/sample/src/sample_pipeline_object.cpp`

closed 2019-01-31 by fiorano10

songshan0321, 2019-12-08:
I have written a easy-to-use ROS wrapper for OpenVINO R3 Inference Engine, with a step-by-step instruction! =) https://github.com/songshan0321/ros_vino