intel/ros_openvino_toolkit #10: YoloV2 making random bounding boxes

fiorano10, 2019-02-08:
After making a couple of changes, I don't get any errors but once the stream starts it just makes random bounding boxes like [this](https://drive.google.com/file/d/1AQJyqIAJlFPl8BCxaD0WbpftXkLIr54U/view?usp=sharing). Here is the [model](https://drive.google.com/drive/folders/1n5xjVWplQn85yj81j2aZUTIb65yQyEGA?usp=sharing) I'm using. It works fine out of the ROS wrapper,  I can even run it on the GPU example [here](https://drive.google.com/file/d/1MCi5-QiRYxZ27H4qVZm-qiwvLiRgVCAB/view?usp=sharing)

chaoli2, 2019-02-28:
@fiorano10 
I'm reproducing your env in recent days. Now I'm able to generate the same YOLO IR (identical md5sum) with yours from the original [yolo weights](https://pjreddie.com/media/files/yolov2-voc.weights). I also reproduce the random bounding boxes issue as you did. But I could only  run the detection with GPU as inference engine. When I attempted to run the detection with CPU, I got an "std::exception" from OpenVINO interface. Have you ever encountered the same error as shown below?

The reason for random bounding boxes in ROS wrapper is probably because the lack of support. We were not intended to support YOLO when we design the wrapper at the beginning. So there is no class to parse and display detection results from YOLO. We are now evaluating the effort to support YOLO. 

[image]

fiorano10, 2019-02-28:
I haven’t tried running YOLO on CPU, so I’m not sure about that. I can test that it tomorrow. Is there a way to make YOLO work with the current ROS wrapper? What would I need to do? How much time will it take for the ROS wrapper to officially support YOLO?

Thanks a lot for your help!

chaoli2, 2019-02-28:
@fiorano10 
I think the main gap for YOLO support is the result parsing function. Currently, we only considered the parsing of SSD [in this class](https://github.com/intel/ros_openvino_toolkit/blob/devel/dynamic_vino_lib/src/inferences/object_detection.cpp). Besides that, the [model loader class](https://github.com/intel/ros_openvino_toolkit/blob/devel/dynamic_vino_lib/src/models/object_detection_model.cpp) (which you modified) and [result display class](https://github.com/intel/ros_openvino_toolkit/blob/master/dynamic_vino_lib/src/outputs/image_window_output.cpp) also need some efforts to update. 

The accurate time taken depends on the root cause why CPU cannot run YOLO detection. If your requirement is just running yolov2-voc detection with GPU, the prioritized task should be the result parsing and the display module. So I'd like to suggest you show us your explicit goal. Then my team will discuss to see  how to support :)

fiorano10, 2019-02-28:
Our requirement is just to run yolov2-voc on the GPU. The explicit goal is to detect people in an unstructured environment. We need to use infrared cameras due to their wider FOV, and YOLO seesm to work with the IR stream as well.

fiorano10, 2019-03-18:
@chaoli2 Any update on the Yolov2 issue?

chaoli2, 2019-03-19:
@fiorano10 
The Yolov2 porting is almost finished. The result parsing modules is done. All that's left is the display module and the final integration test. I expect that the patch will be ready in 1~2 days.

chaoli2, 2019-03-21:
@fiorano10
Here is the [temporary branch](https://github.com/chaoli2/ros_openvino_toolkit-1/tree/yolov2_support) for YOLOv2 support. All function is ready. You can also refer to this [commit](https://github.com/chaoli2/ros_openvino_toolkit-1/commit/073ba1b0134c4c9eeea09bb588380c97a9145ccb) for your customization
Since there are some conflicts with original SSD300 framework, we have to do some refactor before merging it into our main branch (intel-devel). I think it will be finished in few working days.

closed 2019-03-26 by fiorano10