intel/ros_openvino_toolkit #11: RealSenseTopic For pipeline people launch doesn't work

fiorano10, 2019-02-13:
When trying to run people detection with RealSenseTopic instead of launching the RealSense instance, I get the following errors:

`[ WARNING ] No data received in CameraTopic instance`
`[ WARNING ] Failed to get frame from input_device.`

chaoli2, 2019-02-22:
@fiorano10 
This warning is thrown [here](https://github.com/intel/ros_openvino_toolkit/blob/122f17826e2bfc699172094b1de24a826bca4763/dynamic_vino_lib/src/inputs/realsense_camera_topic.cpp#L59), meaning that no data was read from the camera topic "/camera/color/image_raw". Will you confirm the topic and data stream status with:
```
rostopic list                          #expect that /camera/color/image_raw exists
rostopic echo /camera/color/image_raw  #expect that camera data are camptured
```

BTW, may I know how you run your RealSense node? With this [project](https://github.com/intel-ros/realsense)(and which tag) or other methods?

fiorano10, 2019-02-25:
I am using the realsense ros wrapper that you mentioned. I can confirm that the messages are being published to that topic.

fiorano10, 2019-02-28:
Are you able to run RealSenseCameraTopic for emotion and face detection?

chaoli2, 2019-03-07:
@fiorano10 
I've merge this [PR ](https://github.com/intel/ros_openvino_toolkit/pull/16)to resolve this issue. Will you test that again? Thanks.

fiorano10, 2019-03-07:
@chaoli2 
The PR does work, but once you try to exit the node 'CNTRL + C' then we get the same warnings before the node actually shuts down.

`[ WARNING ] Failed to get frame from input_device.`
`[ WARNING ] No data received in CameraTopic instance.`

chaoli2, 2019-03-08:
@fiorano10 
Thanks for your feedback. I submitted this [PR ](https://github.com/intel/ros_openvino_toolkit/pull/18)to fix it.

fiorano10, 2019-03-08:
@chaoli2 Thanks so much!

closed 2019-03-08 by fiorano10