intel/ros_openvino_toolkit #25: Any interest for RaspberryPi install?

droter, 2019-03-28:
I noticed OpenVino will now run on a raspberry pi.

Is there any interest in updating this package to run with Ubuntu 16.04/Kinetic on the RPi?

Thanks,

Matt

chaoli2, 2019-03-29:
@droter 
Thanks for your interests. Currently we've not planned to support RPi in this package.
Not sure whether you've seen this official OpenVINO [installation guide](https://software.intel.com/en-us/articles/OpenVINO-Install-RaspberryPI) for RPi. I go through it and assume that the installation steps are verified by OpenVINO owners. ROS Kinetic is also supported on RPi, as shown [here](http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi). 
So I suggest you try to integrate them on your hardware and give us feedback. We will try to support but not guaranteed due to limited resource. Patches from the community to enable RPi are highly welcome in our package.

droter, 2019-03-29:
I have been able to get OpenVino running with Raspian Stretch.  I use Ubuntu on the RPi and the install will not work.  Hopefully they will add Ubuntu to the supported OS's. Your package looks like fun to use with ROS.

closed 2019-04-23 by chaoli2