intel/ros_openvino_toolkit #26: Running object detection from cameratopic

NickMcKillip, 2019-04-17:
We are trying to run object detection from a camera topic. It is not from a realsense camera, but is structured as an image message.
And you documentation says "any ROS topic which is structured in image message"
I am getting the following errors.
[ WARNING ] No data received in CameraTopic instance
[ WARNING ] Failed to get frame from input_device.

chaoli2, 2019-04-18:
@NickMcKillip 
Shall we know which branch you are using? And what is your camera topic's name?

BTW, We notice that currently there is some hard code in our project, when subscribing camera topic [here](https://github.com/intel/ros_openvino_toolkit/blob/fcf0f9d38cfca40ad3b08150c28e43e5537bc54d/dynamic_vino_lib/src/inputs/realsense_camera_topic.cpp#L40). We'll evaluate and submit topic remapping patch in future.

NickMcKillip, 2019-04-18:
I'm using the master branch. The topics name is /front_camera/image_raw. I already noticed the hard coding as well and already changed it to our topic name, but still got the error.

chaoli2, 2019-04-19:
@NickMcKillip 
Which launch file are you using? Have you update corresponding yaml like [this one](https://github.com/intel/ros_openvino_toolkit/blob/fcf0f9d38cfca40ad3b08150c28e43e5537bc54d/vino_launch/param/pipeline_object.yaml#L3)? Will you please change ```inputs: [RealSenseCamera]``` to ```inputs: [RealSenseCameraTopic]``` and have try.

matteopantano, 2019-06-05:
@NickMcKillip 
@chaoli2 

Were you able to solve?

If I try to run one model I have the error:
`[ WARNING ] Failed to get frame from input_device`

My .yaml looks like the following:
```
Pipelines:
- name: object
  inputs: [StandardCamera]
  infers:
    - name: ObjectDetection
      model: /home/matteo/openvino_models/person-detection-retail-0013.xml
      engine: CPU
      label: to/be/set/xxx.labels
      batch: 1
      confidence_threshold: 0.5
      enable_roi_constraint: true # set enable_roi_constraint to false if you don't want to make the inferred ROI (region of interest) constrained into the camera frame
    - name: PersonReidentification
      model: /home/matteo/openvino_models/person-reidentification-retail-0076.xml
      engine: CPU
      label: to/be/set/xxx.labels
      batch: 1
      confidence_threshold: 0.7
  outputs: [ImageWindow, RosTopic, RViz]
  connects:
    - left: StandardCamera
      right: [ObjectDetection]
    - left: ObjectDetection
      right: [PersonReidentification]
    - left: PersonReidentification
      right: [ImageWindow, RosTopic, RViz]

Common:
  camera_topic: /usb_cam/image_raw
```

chaoli2, 2019-06-06:
@matteopantano 
We're sorry that the input topic is hard code [here](https://github.com/intel/ros_openvino_toolkit/blob/7d7aa11a457e90431dc2cd6ff101ff458fe2ec1f/dynamic_vino_lib/src/inputs/realsense_camera_topic.cpp#L40) at this moment. We'll fix it in next release,

matteopantano, 2019-06-08:
As long I was more concerned with using a standard usb camera instead of the realsense I have solved changing the opencv index [here](https://github.com/intel/ros_openvino_toolkit/blob/7d7aa11a457e90431dc2cd6ff101ff458fe2ec1f/dynamic_vino_lib/src/inputs/standard_camera.cpp#L26). Hence, it is not necessary to launch the camera topic from another node as long this function access to the camera by itself. Hope it helps for now!

@chaoli2 
I think that in the next release you should change this [section](https://github.com/intel/ros_openvino_toolkit/blob/7d7aa11a457e90431dc2cd6ff101ff458fe2ec1f/dynamic_vino_lib/src/inputs/standard_camera.cpp#L26-L50) as long now you are not subscribing to any topic and you are just opening the camera with OpenCV, that may lead confusion to the users.

songshan0321, 2019-11-19:
I have written a ROS wrapper for OpenVINO. It is pretty straight forward to use.
https://github.com/songshan0321/ros_vino