intel/ros_openvino_toolkit #32: Running with Yolov3 models

amkurup, 2019-04-25:
Hi,
I am trying to run this package with yolo-v3 as given in [this tutorial](https://docs.openvinotoolkit.org/latest/_docs_MO_DG_prepare_model_convert_model_tf_specific_Convert_YOLO_From_Tensorflow.html) from intel. I have faced no problem running on the CPU with the ssd300 model downloaded from intel.

But when I try the the sample on the coco dataset, the program quits with this error: 
`[ERROR] object detection network should have only one output`. 
As I understand, the ssd300 model has 21 outputs. If so, what could be causing this issue?

The files I have tried are exactly as mentioned in the link above: [yolov3.weights](https://pjreddie.com/media/files/yolov3.weights) file with [coco.names](https://raw.githubusercontent.com/pjreddie/darknet/master/data/coco.names).
I first used tensorflow-yolov3 to convert to frozen model using:
`python3 convert_weights_pb.py --class_names coco.names --data_format NHWC --weights_file yolov3.weights`
and then converted to IR using:
`python3 mo_tf.py --input_model /path/to/yolo_v3.pb --tensorflow_use_custom_operations_config $MO_ROOT/extensions/front/tf/yolo_v3.json`

My system:
Ubuntu16.04 & Ros Kinetic
ros_openvino_toolkit commit: fcf0f9d
dldt checkout: `2018_R5` 
open_model_zoo: `2018_R5`

amkurup, 2019-04-25:
Not sure if this is relevant but I tried adding `--tensorflow_object_detection_api_pipeline_config pipeline.config` as mentioned in issue (https://github.com/intel/ros_openvino_toolkit/issues/6#issuecomment-457992911) but I do not have a pipeline.config file.

chaoli2, 2019-04-28:
@amkurup 
This exception is thrown [here](https://github.com/intel/ros_openvino_toolkit/blob/24bbdc6252b797964ddcb31505fb809d99833cf3/dynamic_vino_lib/src/models/base_model.cpp#L87). Seems that the output size here is not the network output size, but the pipeline output. See the [flow chart](https://github.com/intel/ros_openvino_toolkit/blob/24bbdc6252b797964ddcb31505fb809d99833cf3/dynamic_vino_lib/src/models/base_model.cpp#L87) for more details. I'd like to try to reproduce your issue from my side and will let you know if there is any update.

amkurup, 2019-04-28:
Hi @chaoli2 
I was able to trace the exception to its source, but as you know, the basic yolo-v3 model has 3 output yolo layers. Seems like they would always throw such an exception! Yet, there are users who claim this toolkit to be working with yolo-v3 generated models. Could you please elaborate the workaround?

I am guessing I am using mo_tf.py in the wrong way. My usage is: 
`python3 mo_tf.py --input_model /path/to/yolo_v3.pb --tensorflow_use_custom_operations_config $MO_ROOT/extensions/front/tf/yolo_v3.json`

chaoli2, 2019-04-29:
@amkurup 
The command we use to convert is:
```
python3 mo_tf.py --input_model /path/to/frozen_darknet_yolov3_model.pb --tensorflow_use_custom_operations_config 
$MO_ROOT/extensions/front/tf/yolo_v3.json --input_shape=[1,416,416,3]
```

After that, we've tried the generated model with OpenVINO native sample code _(/opt/intel/computer_vision_sdk/inference_engine/samples/object_detection_demo_yolov3_async)_ and it works well, but not tested in this ROS package yet.

amkurup, 2019-04-29:
Thanks @chaoli2 
I will try to use the above commands and the sample application as a test.
I am eager to find out your results with the ROS package.

amitrege, 2019-06-11:
@amkurup @chaoli2 Has this been resolved yet?

I am using this command `python3 mo_tf.py --input_model /path/to/frozen_darknet_yolov3_model.pb --tensorflow_use_custom_operations_config 
$MO_ROOT/extensions/front/tf/yolo_v3.json --input_shape=[1,416,416,3]` to generate IR.

Then I make a change to the pipeline_object_topic.yaml file and change the model parameter to point to the generated .xml file.

I get the same error: `Object Detectionnetwork should have only one input`

amkurup, 2019-06-11:
@amitrege As far as my knowledge goes, seems like @chaoli2 and team will have to look into this. The fundamental issue is the use of Yolov3 (since it has 3 output layers). For my project, I was able to use OpenVINO's demo code for Yolov3 and bind it with this package (for ROS support)

amitrege, 2019-06-12:
Thanks @amkurup . Is your code open source by any chance?