intel/ros_openvino_toolkit #41: Undefined Symbol error

amitrege, 2019-06-03:
I seem to get the following error
` 
/home/amit/dev-work/openvino_ws/devel/lib/dynamic_vino_sample/pipeline_with_params: symbol lookup error: /opt/intel/openvino_2019.1.144/deployment_tools/inference_engine/lib/intel64/libinference_engine.so: undefined symbol: _ZN3tbb10interface78internal15task_arena_base24internal_max_concurrencyEPKNS0_10task_arenaE
[pipeline_with_params-1] process has died [pid 24368, exit code 127, cmd /home/amit/dev-work/openvino_ws/devel/lib/dynamic_vino_sample/pipeline_with_params /openvino_toolkit/detected_objects:=/ros_openvino_toolkit/detected_objects /openvino_toolkit/images:=/ros_openvino_toolkit/image_rviz __name:=pipeline_with_params __log:=/home/amit/.ros/log/173eb096-862e-11e9-a084-080027a99f6a/pipeline_with_params-1.log].
log file: /home/amit/.ros/log/173eb096-862e-11e9-a084-080027a99f6a/pipeline_with_params-1*.log
`
when I try to run `roslaunch vino_launch pipeline_object.launch`.

I am running Ubuntu 18.04. My OpenVINO installation and RealSense camera both seem to be working fine so I am not quite sure what this error is about.

RachelRen05, 2019-06-04:
@amitrege 
From your log, the openvino you installed is the version of 2019R1.1, we are going to support this version in our project.

Your problem may be caused by the source of the wrong environment variable.
If you refer to the [guide](https://docs.openvinotoolkit.org/latest/_docs_install_guides_installing_openvino_linux.html), and install it as root, the following steps need to be executed before launch.

` source /opt/intel/openvino/bin/setupvars.sh`
` export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/opt/intel/openvino/deployment_tools/inference_engine/samples/build/intel64/Release/lib`

If you have any questions, feel free to let us know.

amitrege, 2019-06-04:
Thanks for your response.

>From your log, the openvino you installed is the version of 2019R1.1, we are going to support this version in our project.

Does this mean that this version is not supported at the moment? If so, what version is supported currently?

>source /opt/intel/openvino/bin/setupvars.sh
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/opt/intel/openvino/deployment_tools/inference_engine/samples/build/intel64/Release/lib

I have done this already but I still get the same error.

RachelRen05, 2019-06-05:
@amitrege 
The version we are currently supporting is 2018 R5, you can refer to our [guide](https://github.com/intel/ros_openvino_toolkit/blob/devel/doc/BINARY_VERSION_README.md#3-environment-setup) to set up your environment. I've also just seen that our tutorial needs an update for OpenVINO toolkit version,

> **Choose 2018R5 when download tarball**

we'll work on that next. Thanks.

closed 2020-02-20 by LewisLiuPub