intel/ros_openvino_toolkit #42: Cannot build ros_openvino error expected constructor

matteopantano, 2019-06-03:
To whom it may concern,

I am trying to build the package. I have followed the guide but unfortunately I encounter a build error when compiling the samples.

```
_________________________________________________________________________________________________________________________________
Errors     << dynamic_vino_sample:make /home/matteo/rovitis_ws/logs/dynamic_vino_sample/build.make.001.log                       
In file included from /home/matteo/rovitis_ws/src/ros_openvino_toolkit/sample/src/image_segmentation_server.cpp:60:0:
/home/matteo/rovitis_ws/src/ros_openvino_toolkit/sample/include/sample/utility.hpp:159:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_uint32(n_ag, 16, num_batch_ag_message);
              ^
/home/matteo/rovitis_ws/src/ros_openvino_toolkit/sample/include/sample/utility.hpp:162:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_uint32(n_hp, 16, num_batch_hp_message);
```
Any idea why I get this error?

chaoli2, 2019-06-04:
@matteopantano
May I know which branch or tag of this project are you using? What is your OpenVINO's branch or tag?
I'd like to reproduce this issue from my side.

matteopantano, 2019-06-04:
I am using openVINO R1 build April 2019 and the master branch from this project

chaoli2, 2019-06-04:
@matteopantano 
There may be some issues for OpenVINO R1 support. We're now still working to upgrade OpenVINO support for 2018 R5 to 2019 R1. If you'd like to use this project at this moment, I suggest roll back to 2018 R5 temporarily. 

If you still have any build issue while using 2018 R5, please let me know.

matteopantano, 2019-06-04:
From Intel website the Tareball OpenVINO R5 2018 is not available for download. What should I do?

JimWest, 2019-06-04:
> 
> 
> @chaoli2
> From Intel website the Tareball OpenVINO R5 2018 is not available for download. What should I do?
> Should I try the OpenSource version branch 2018?

I've got the exact same error with OpenVINO R5 2018, but I tried to build it on Debian 9 (even  if it's not supported, everythings working except the ros sample package).

chaoli2, 2019-06-04:
@matteopantano 
Here is the [link](http://registrationcenter-download.intel.com/akdlm/irc_nas/15078/l_openvino_toolkit_p_2018.5.455.tgz) of OpenVINO 2018 R5.
You are also suggested to double check the gflags issue mentioned by JimWest.

@JimWest 
Thanks for providing such valuable information.

matteopantano, 2019-06-04:
@chaoli2
I have also installed the libgflags-dev, might be my issue here.

I will check and let you know.

matteopantano, 2019-06-04:
I tried as suggested but I have now another error:

```
CMakeFiles/image_segmentation_server.dir/src/image_segmentation_server.cpp.o: In function `parseAndCheckCommandLine(int, char**)':
image_segmentation_server.cpp:(.text+0x3164): undefined reference to `google::ParseCommandLineNonHelpFlags(int*, char***, bool)'
CMakeFiles/image_segmentation_server.dir/src/image_segmentation_server.cpp.o: In function `__static_initialization_and_destruction_0(int, int)':
image_segmentation_server.cpp:(.text+0x3727): undefined reference to `google::FlagRegisterer::FlagRegisterer<bool>(char const*, char const*, char const*, bool*, bool*)'
image_segmentation_server.cpp:(.text+0x37a2): undefined reference to `google::FlagRegisterer::FlagRegisterer<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(char const*, char const*, char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
```
@JimWest did you follow [this](https://github.com/gflags/gflags/blob/master/INSTALL.md) for
installing Gflags from source?

JimWest, 2019-06-04:
> 
> 
> I tried as suggested but I have now another error:
> 
> ```
> Errors     << dynamic_vino_sample:make /home/matteo/rovitis_ws/logs/dynamic_vino_sample/build.make.011.log
> CMakeFiles/image_segmentation_server.dir/src/image_segmentation_server.cpp.o: In function `parseAndCheckCommandLine(int, char**)':
> image_segmentation_server.cpp:(.text+0x3164): undefined reference to `google::ParseCommandLineNonHelpFlags(int*, char***, bool)'
> CMakeFiles/image_segmentation_server.dir/src/image_segmentation_server.cpp.o: In function `__static_initialization_and_destruction_0(int, int)':
> image_segmentation_server.cpp:(.text+0x3727): undefined reference to `google::FlagRegisterer::FlagRegisterer<bool>(char const*, char const*, char const*, bool*, bool*)'
> image_segmentation_server.cpp:(.text+0x37a2): undefined reference to `google::FlagRegisterer::FlagRegisterer<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >(char const*, char const*, char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >*)'
> ```
> 
> @JimWest did you follow [this](https://github.com/gflags/gflags/blob/master/INSTALL.md) for installing Gflags from source?

Looks like it's still linking the old version. I've installed a new one in the system, thats why its working for me.

The problem is that Gflags isn't added properly to the CMakeLists.txt from this repo, it only looks at strange include paths or in the default path, and tries to link it via an environment variable. It would be much better if this would use the standard cmake "find_package" function, and link that library and include that IncludeDirs.

matteopantano, 2019-06-04:
I have solved after building several times with a `catkin clean` in my workspace

closed 2019-06-04 by matteopantano