intel/ros_openvino_toolkit #43: Error reading network: cannot parse future versions: 5

matteopantano, 2019-06-04:
After successfully having compiled the system with openVINO R5 2018 and [this](https://github.com/gflags/gflags) gflags. I do now have the following:

```
[ INFO ] Loading network files
[ ERROR ] Error reading network: cannot parse future versions: 5
/teamcity/work/scoring_engine_build/releases_2018_R5/src/inference_engine/ie_cnn_net_reader_impl.cpp:144
/opt/intel/computer_vision_sdk_2018.5.455/deployment_tools/inference_engine/include/details/ie_exception_conversion.hpp:71
[pipeline_with_params-1] process has died [pid 11922, exit code 1, cmd /home/matteo/rovitis_ws/devel/lib/dynamic_vino_sample/pipeline_with_params /openvino_toolkit/reidentified_persons:=/ros_openvino_toolkit/reidentified_persons /openvino_toolkit/images:=/ros_openvino_toolkit/image_rviz __name:=pipeline_with_params __log:=/home/matteo/.ros/log/7b569c70-86e6-11e9-ac4f-f8597154e5bb/pipeline_with_params-1.log].
log file: /home/matteo/.ros/log/7b569c70-86e6-11e9-ac4f-f8597154e5bb/pipeline_with_params-1*.log
```
I would like to point out that I have downloaded the `.xml` from [this](https://download.01.org/opencv/) which is a suggested option by openVINO documentation [here](http://docs.openvinotoolkit.org/latest/_docs_Pre_Trained_Models.html)

chaoli2, 2019-06-05:
@matteopantano 
Version 5 means that the downloaded model is generated with OpenVINO 2019 R1, which is not compatible with OpenVINO 2018 R5 (version = "4"). 
Here is an [example](https://download.01.org/opencv/2019/open_model_zoo/R1/models_bin_20190329_105530/face-detection-adas-0001/FP32/face-detection-adas-0001.xml) to show this. Please note the first line of the .xml, in which version = 5.

In this [page](https://download.01.org/opencv/2019/open_model_zoo/R1/), there are two folders providing "version 4" and "version 5" respectively. I notice that directory "models_bin" provides "version 4" and "models_bin_20190329_105530" provides "version 5" for the same model. Will you please re-download your model to confirm that "version = 4" at the first line of the .xml?

matteopantano, 2019-06-05:
I have tried the models of version 4. Now I have the following error:

[ INFO ] max proposal count is: 200
[ ERROR ] Incorrect number of input edges for layer mbox_conf1/out/conv/flat/reshape
/teamcity/work/scoring_engine_build/releases_2018_R5/src/mkldnn_plugin/nodes/mkldnn_reshape_node.cpp:19
/opt/intel/computer_vision_sdk_2018.5.455/deployment_tools/inference_engine/include/details/ie_exception_conversion.hpp:71
[pipeline_with_params-1] process has died [pid 5036, exit code 1, cmd /home/matteo/rovitis_ws/devel/lib/dynamic_vino_sample/pipeline_with_params /openvino_toolkit/reidentified_persons:=/ros_openvino_toolkit/reidentified_persons /openvino_toolkit/images:=/ros_openvino_toolkit/image_rviz __name:=pipeline_with_params __log:=/home/matteo/.ros/log/8c3ae5a0-8740-11e9-b2ad-5d72cb9bb797/pipeline_with_params-1.log].
log file: /home/matteo/.ros/log/8c3ae5a0-8740-11e9-b2ad-5d72cb9bb797/pipeline_with_params-1*.log

matteopantano, 2019-06-05:
Sorry, closed the issue for error

chaoli2, 2019-06-05:
Which model are you using?

matteopantano, 2019-06-05:
These two:

```
models_bin/person-detection-retail-0013/FP32/person-detection-retail-0013.xml
models_bin/person-reidentification-retail-0076/FP32/person-reidentification-retail-0076.xml
```

chaoli2, 2019-06-05:
@matteopantano 

I cannot reproduce your issue from my side. Everything works well.
Let's first double check the md5sum of these two IRs to guarantee they are identical.

a0351077bf72d5d0af6b9c180ce5a996  person-detection-retail-0013.bin
fd62a89f0cebd28e84b5bea60f7105ff  person-detection-retail-0013.xml

36be1a27d1753fc3b5b47688aa1e1044  person-reidentification-retail-0076.bin
a3e6fe6d4435720ad8f3a907dde3e92c  person-reidentification-retail-0076.xml

matteopantano, 2019-06-05:
@chaoli2 

These are mine:

e09bb33802c088aaeb10f23373cb4433  person-detection-retail-0013.bin
027a4ecc7469378742e5afabbe5628c8  person-detection-retail-0013.xml

**36be1a27d1753fc3b5b47688aa1e1044  person-reidentification-retail-0076.bin**
5e72ad39f32ed9ef45c89fde500a0938  person-reidentification-retail-0076.xm

Could you share your models?

matteopantano, 2019-06-05:
I have solved via downloading the models as described in the wiki

closed 2019-06-05 by matteopantano