intel/ros_openvino_toolkit #44: Can't find myriad

satchelh, 2019-06-15:
Whenever I launch the pipeline_people_myriad.launch file in the vino_launch package, I get the following output:

Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://arl-superpc:39025/

SUMMARY
========

PARAMETERS
 * /pipeline_with_params/param_file: /home/satchel/cat...
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /rviz/default_rviz: /home/satchel/cat...

NODES
  /
    pipeline_with_params (dynamic_vino_sample/pipeline_with_params)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [7641]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b468c3ae-8f14-11e9-b7ea-f894c2575b6b
process[rosout-1]: started with pid [7654]
started core service [/rosout]
process[pipeline_with_params-2]: started with pid [7657]
process[rviz-3]: started with pid [7658]
[ INFO ] FLAGS_config=/home/satchel/catkin_ws/src/ros_openvino_toolkit/vino_launch/param/pipeline_people_myriad.yaml
InferenceEngine: 
	API version ............ 1.4
	Build .................. custom_HEAD_91160e65cf66e7fbc709245a8ba948a4c555e443
[ INFO ] Pipeline size: 1
[ INFO ] Inferences size: 4
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ INFO ] Inference Params:name=FaceDetection
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ INFO ] Inference Params:name=AgeGenderRecognition
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ INFO ] Inference Params:name=EmotionRecognition
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ INFO ] Inference Params:name=HeadPoseEstimation
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ INFO ] Pipeline Params:name=people
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ INFO ] --------parameters DUMP---------------------
[ INFO ] Pipeline: people
[ INFO ] 	Inputs: StandardCamera, 
[ INFO ] 	Outputs: ImageWindow, RosTopic, RViz, 
[ INFO ] 	Inferences: 
[ INFO ] 		Name: FaceDetection
[ INFO ] 		Model: /opt/intel/computer_vision_sdk/deployment_tools/intel_models/face-detection-adas-0001/FP16/face-detection-adas-0001.xml
[ INFO ] 		Engine: MYRIAD
[ INFO ] 		Label: to/be/set/xxx.labels
[ INFO ] 		Batch: 1
[ INFO ] 		Confidence_threshold: 0.5
[ INFO ] 		Enable_roi_constraint: 0
[ INFO ] 		Name: AgeGenderRecognition
[ INFO ] 		Model: /opt/intel/computer_vision_sdk/deployment_tools/intel_models/age-gender-recognition-retail-0013/FP16/age-gender-recognition-retail-0013.xml
[ INFO ] 		Engine: MYRIAD
[ INFO ] 		Label: to/be/set/xxx.labels
[ INFO ] 		Batch: 16
[ INFO ] 		Confidence_threshold: 0.5
[ INFO ] 		Enable_roi_constraint: 0
[ INFO ] 		Name: EmotionRecognition
[ INFO ] 		Model: /opt/intel/computer_vision_sdk/deployment_tools/intel_models/emotions-recognition-retail-0003/FP16/emotions-recognition-retail-0003.xml
[ INFO ] 		Engine: MYRIAD
[ INFO ] 		Label: to/be/set/xxx.labels
[ INFO ] 		Batch: 16
[ INFO ] 		Confidence_threshold: 0.5
[ INFO ] 		Enable_roi_constraint: 0
[ INFO ] 		Name: HeadPoseEstimation
[ INFO ] 		Model: /opt/intel/computer_vision_sdk/deployment_tools/intel_models/head-pose-estimation-adas-0001/FP16/head-pose-estimation-adas-0001.xml
[ INFO ] 		Engine: MYRIAD
[ INFO ] 		Label: to/be/set/xxx.labels
[ INFO ] 		Batch: 16
[ INFO ] 		Confidence_threshold: 0.5
[ INFO ] 		Enable_roi_constraint: 0
[ INFO ] 	Connections: 
[ INFO ] 		AgeGenderRecognition->ImageWindow
[ INFO ] 		AgeGenderRecognition->RosTopic
[ INFO ] 		AgeGenderRecognition->RViz
[ INFO ] 		EmotionRecognition->ImageWindow
[ INFO ] 		EmotionRecognition->RosTopic
[ INFO ] 		EmotionRecognition->RViz
[ INFO ] 		FaceDetection->AgeGenderRecognition
[ INFO ] 		FaceDetection->EmotionRecognition
[ INFO ] 		FaceDetection->HeadPoseEstimation
[ INFO ] 		FaceDetection->ImageWindow
[ INFO ] 		FaceDetection->RosTopic
[ INFO ] 		FaceDetection->RViz
[ INFO ] 		HeadPoseEstimation->ImageWindow
[ INFO ] 		HeadPoseEstimation->RosTopic
[ INFO ] 		HeadPoseEstimation->RViz
[ INFO ] 		StandardCamera->FaceDetection
[ INFO ] Common:
[ INFO ] 	camera_topic: 
[ INFO ] 	custom_cpu_library: 
[ INFO ] 	custom_cldnn_library: 
[ INFO ] 	enable_performance_count: 0
[ INFO ] Parsing InputDvice: StandardCamera
[ INFO] [1560565470.693783254]: rviz version 1.12.17
[ INFO] [1560565470.693810182]: compiled against Qt version 5.5.1
[ INFO] [1560565470.693817283]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO ]  ... Adding one Input device: StandardCamera
[ INFO ] Parsing Output: ImageWindow
[ INFO ]  ... Adding one Output: ImageWindow
[ INFO ] Parsing Output: RosTopic
[ INFO ]  ... Adding one Output: RosTopic
[ INFO ] Parsing Output: RViz
[ INFO ]  ... Adding one Output: RViz
[ INFO ] Parsing Inference: FaceDetection
[ ERROR ] Cannot find plugin to use :

/home/satchel/code/dldt/inference-engine/include/ie_plugin_dispatcher.hpp:86
[pipeline_with_params-2] process has died [pid 7657, exit code 1, cmd /home/satchel/catkin_ws/devel/lib/dynamic_vino_sample/pipeline_with_params /openvino_toolkit/faces:=/ros_openvino_toolkit/face_detection /openvino_toolkit/emotions:=/ros_openvino_toolkit/emotions_recognition /openvino_toolkit/headposes:=/ros_openvino_toolkit/headposes_estimation /openvino_toolkit/age_genders:=/ros_openvino_toolkit/age_genders_Recognition /openvino_toolkit/images:=/ros_openvino_toolkit/image_rviz __name:=pipeline_with_params __log:=/home/satchel/.ros/log/b468c3ae-8f14-11e9-b7ea-f894c2575b6b/pipeline_with_params-2.log].
log file: /home/satchel/.ros/log/b468c3ae-8f14-11e9-b7ea-f894c2575b6b/pipeline_with_params-2*.log
[ INFO] [1560565471.169292058]: Stereo is NOT SUPPORTED
[ INFO] [1560565471.169362763]: OpenGl version: 4.6 (GLSL 4.6).

However, the pipeline_people_oss.launch works fine with my usb camera. The intel stick works, it shows up as device when running lsusb and I have used it before. How can I fix this?

chaoli2, 2019-06-18:
From your log, I think you are using the dldt (the open source version of OpenVINO). However, the myriad plugin is only available in binary version of OpenVINO. Please see [here ](https://github.com/intel/ros_openvino_toolkit#enable-intel-neural-compute-stick-2-intel-ncs-2-under-the-openvino-open-source-version-optional-)for installation steps.

satchelh, 2019-06-18:
> From your log, I think you are using the dldt (the open source version of OpenVINO). However, the myriad plugin is only available in binary version of OpenVINO. Please see [here [...]

This solved my issue, thanks!

closed 2019-06-18 by satchelh