intel/ros_openvino_toolkit #45: Support for Intel® Movidius™ Neural Compute Stick

matteopantano, 2019-06-16:
In the example config file is specified that only CPU and GPU are currently supported by the repository. When the support for the Neural Compute Stick will roll out?

Best regards,

Matteo

chaoli2, 2019-06-18:
@matteopantano 
Movidius Neural Compute Stick is already supported. See [guide ](https://github.com/intel/ros_openvino_toolkit#enable-intel-neural-compute-stick-2-intel-ncs-2-under-the-openvino-open-source-version-optional-)here for setup.

Here is the example [config file](https://github.com/intel/ros_openvino_toolkit/blob/master/vino_launch/param/pipeline_people_myriad.yaml).

sradmard, 2020-02-06:
Hi,
I have a question regarding the support for the original Intel® Movidius™ NCS (not the Intel® NCS 2). From this [link](https://software.intel.com/en-us/articles/transitioning-from-intel-movidius-neural-compute-sdk-to-openvino-toolkit), it is stated that the OpenVINO supports the original NCS. I am also able to run the demos of OpenVINO with my original NCS. However, when I run the ros_openvino_toolkit Myriad examples, on my original NCS, I get the following error:
`[ ERROR ] Can not init Myriad device: NC_ERROR`

However, I can easily run the Myriad examples using my NCS 2. 
My question is, does this package support the original Movidius NCS? 
Thank you in advance.

LewisLiuPub, 2020-02-20:
@sradmard Hi, how about the latest status? did you make NCS work with ROS OpenVINO toollkit?

sradmard, 2020-02-20:
Hi @LewisLiuPub, it is quite wired as it runs randomly. Sometimes it runs an example launch file, but some other times the same launch file does return the following error:
`E: [ncAPI] [    429093] [pipeline_with_p] ncDeviceOpen:859	Device doesn't appear after boot
[ ERROR ] Can not init Myriad device: NC_ERROR`

Every time after this error, when I check the lsusb, it looks like the system does not recognize the NCS anymore. I have to re-plug the NCS hardware to get the following entry for lsusb:
`Bus 001 Device 025: ID 03e7:2150 `

I have no issue with my NCS 2, it is only my original NCS that causes these problems. Any observation from your side?

LewisLiuPub, 2020-02-21:
@sradmard Yes, we have also met such issues before, NCS is not as stable as NCS2 after NCS2 related driver/plugins installed.
I guess it is probably produced from the leverage of NCS2 related package for NCS's functionalities. NCS has big differences in HW layer comparing with NCS2, but they shares the same interfaces in OpenVINO.
The workaround currently is to re-plug the NCS device, or directly use NCS2, rather than NCS.

MaajidKhan, 2020-08-16:
Hello @sradmard. 
The issue is that the device you are using is the original Movidius NCS 1 
ID 03e7:2150 Intel Myriad VPU [Movidius Neural Compute Stick]

The original NCS 1 stick only is supported with the usual ncsdk2 package and till OpenVINO 2020.3.

**Official statement**:
The Intel Movidius Neural Compute Stick (NCS) will be supported in this LTS release, according to LTS policy, NCS stick support will be stopped in the next release (2020.4), but will continue to be available for 2020.3 LTS release updates. 

So, the workaround would be to use the latest NCS2 stick (Movidius NCS 2) for future testing and support purposes.
Bus 001 Device 007: ID 03e7:2485 Intel Movidius MyriadX

The latest NCS2 is compatible with both OV 2020.3 , OV 2020.4 and further Releases.

SiqiWang0806, 2020-09-08:
Hi, may I ask if HDDL is supported in this toolkit?

LewisLiuPub, 2020-09-08:
Currently, ROS wrapper of OpenVINO toolkit doesn't support HDDL. but Intel OpenVINO SDK does.