intel/ros_openvino_toolkit #47: How do I upload my own photos for reidentification?

GalMoore, 2019-08-12:
Where do I add my own photos so that it can recognize me and my friends? I did not manage to see which photos it currently uses and where is the paths for the reidentification?
Thanks!

chaoli2, 2019-08-14:
@GalMoore 
Currently, this project doesn't support reidentification from image input . But there may be workaround to do this: 
1. read your pictures into CV::Mat
2. convert CV::Mat into sensor_msgs/Image.msg with [cv_bridge](https://github.com/ros-perception/vision_opencv)
3. publish converted message into topci "/ros_openvino_toolkit/detected_objects" according to your [launcher configuration](https://github.com/intel/ros_openvino_toolkit/blob/7d7aa11a457e90431dc2cd6ff101ff458fe2ec1f/vino_launch/launch/pipeline_object_topic.launch#L17).
4. launch openvino node with above launch file

@LewisLiuPub @RachelRen05 
Please help review this process and give your comments if any. Thanks.

sradmard, 2020-03-12:
Hi @chaoli2, 
I am also trying to do the same thing, however, unfortunately your suggestion does not work for me. 
First of all, I am trying to modify the "person_reidentification.cpp" so that it can do people identification based on pre-recorded and labelled images, but I am sure I need to modify other files as well. I need to pre-populate the encoding vectors in the "recorded_persons_" in "person_reidentification.cpp" based on pre-recorded images, and that is the easy part. What I am a little bit lost is, where can I create the encoding vectors of my pre-recorded images?
 
Secondly, after encoding my images, I am planning to create a dictionary of the encoding vectors with the keys being the unique labels of people, and then use this dictionary to populate the "recorded_persons_".  Does this sound reasonable? 

At the end, I would appreciate any other tips that might be useful.