intel/ros_openvino_toolkit #48: Unable to run Person_Reidentfication sample

Arnab1311, 2019-08-20:
[pipeline_reidentification_oss.launch] is neither a launch file in package [vino_launch] nor is [vino_launch] a launch file name
The traceback for the exception was written to the log file

This is what i get everytime i try running the sample.

chaoli2, 2019-08-22:
@Arnab1311 
Sorry that ReID features are not available in this package. There are some mistakes in README now. I'll update it soon. We are planning to port ReID features from OpenVINO [ROS2 package](https://github.com/intel/ros2_openvino_toolkit) to this ROS package recently. Thanks for your feedback.

chaoli2, 2019-08-22:
@Arnab1311 
Have you sourced "devel/setup.bash" under your ```catkin_make``` directory?
Checked in source code, [pipeline_reidentification_oss.launch](https://github.com/intel/ros_openvino_toolkit/tree/master/vino_launch/launch) is under [vino_launch/launch](https://github.com/intel/ros_openvino_toolkit/tree/master/vino_launch/launch) folder.

After compilation, the launch file should be copied into "devel" or "install" directory. Would you please check this with a ```find -name "*.launch"``` command?