intel/ros_openvino_toolkit #49: unable build on raspberry pi 3

xlla, 2019-09-28:
catkin_make result:

-- BUILD_SHARED_LIBS is on
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
CMake Error at /home/pi/ros_catkin_ws/install_isolated/share/genmsg/cmake/genmsg-extras.cmake:263 (message):
  Messages depends on unknown pkg: object_msgs (Missing
  'find_package(object_msgs)'?)
Call Stack (most recent call first):
  CMakeLists.txt:48 (generate_messages)

-- Configuring incomplete, errors occurred!
See also "/home/pi/catkin_ws/build_isolated/people_msgs/CMakeFiles/CMakeOutput.log".
See also "/home/pi/catkin_ws/build_isolated/people_msgs/CMakeFiles/CMakeError.log".
<== Failed to process package 'people_msgs': 
  Command '['cmake', '/home/pi/catkin_ws/src/ros_openvino_toolkit/people_msgs', '-DCATKIN_DEVEL_PREFIX=/home/pi/catkin_ws/devel_isolated/people_msgs', '-DCMAKE_INSTALL_PREFIX=/home/pi/catkin_ws/install_isolated', '-G', 'Unix Makefiles']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/pi/catkin_ws/build_isolated/people_msgs && cmake /home/pi/catkin_ws/src/ros_openvino_toolkit/people_msgs -DCATKIN_DEVEL_PREFIX=/home/pi/catkin_ws/devel_isolated/people_msgs -DCMAKE_INSTALL_PREFIX=/home/pi/catkin_ws/install_isolated -G 'Unix Makefiles'

Command failed, exiting.

xlla, 2019-09-28:
I have installed object_msgs.

pi@rpi3:~/catkin_ws $ rosls object_msgs
cmake  msg  package.xml  srv

xlla, 2019-09-28:
another error when build vino_param_lib

[ 50%] Building CXX object CMakeFiles/vino_param_lib.dir/src/param_manager.cpp.o
c++: error: unrecognized command line option ‘-msse4.1’
c++: error: unrecognized command line option ‘-mf16c’
CMakeFiles/vino_param_lib.dir/build.make:62: recipe for target 'CMakeFiles/vino_param_lib.dir/src/param_manager.cpp.o' failed
make[2]: *** [CMakeFiles/vino_param_lib.dir/src/param_manager.cpp.o] Error 1
CMakeFiles/Makefile2:690: recipe for target 'CMakeFiles/vino_param_lib.dir/all' failed
make[1]: *** [CMakeFiles/vino_param_lib.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
<== Failed to process package 'vino_param_lib': 
  Command '['/home/pi/catkin_ws/install_isolated/env.sh', 'make', '-j4', '-l4']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /home/pi/catkin_ws/build_isolated/vino_param_lib && /home/pi/catkin_ws/install_isolated/env.sh make -j4 -l4

Command failed, exiting.

xlla, 2019-09-28:
I think is should be: -march=armv8-a -mtune=cortex-a53 -mfpu=crypto-neon-fp-armv8

chaoli2, 2019-09-29:
@xlla 
Thanks for your feedback. Sorry to say that we have no plan to support ARM-based board yet. So you have to update the compile flags in CMakeLists.txt if you'd like to build this project on raspberry pi 3.

closed 2020-02-20 by LewisLiuPub