intel/ros_openvino_toolkit #50: Implementing my own pipeline

ximenesfel, 2019-10-02:
I would like to implement my own pipeline using other openvino models that were not supported by ros_openvino_toolkit at the moment. Looking at the documentation available at ros_openvino_toolkit github, I'm not able to find a document that shows how to implement my own pipeline (So I need to configure .yaml file to create my pipeline, but I think I need to change more things like creating a new model in dynamic_vino_lib). Do you have this document? If not, can you point me how to do that?

I want to implement the following pipeline:

Object detection
Text detection 
Text recognition

ximenesfel, 2019-10-09:
Thanks, @chaoli2 for your detailed explanation. I will try to implement the new pipeline using your information. If I have any questions I will let you know here.

ximenesfel, 2019-11-20:
Hi.

So I implemented the pipeline and it is working for recognizing only one text (The text detection model detects many ROIs but I only able to run the complete pipeline when existing only one RoI).

I'm using the following [Text Recognition demo](https://github.com/opencv/open_model_zoo/blob/master/demos/text_detection_demo/main.cpp) as a reference to implement this pipeline. 

Bellow my pipeline configuration:

```
Pipelines:
- name: text_recognition
  inputs: [StandardCamera]
  infers:
    - name: TextDetection
      model: text_detection_model path
      engine: CPU
      label: to/be/set/xxx.labels
      batch: 1
    - name: TextRecognition
      model: text_recognition_model path
      engine: CPU
      label: to/be/set/xxx.labels
      batch: 1
  outputs: [ImageWindow]
  confidence_threshold: 0.5
  connects:
    - left: StandardCamera
      right: [TextDetection] 
    - left: TextDetection
      right: [TextRecognition, ImageWindow]
    - left: TextRecognition
      right: [ImageWindow]

OpenvinoCommon:
```
I'm having a problem when I try to obtain the result in fetchResults() method for TextRecognition. Text detection model output the rectangles from each text in the image.

Looking at pipeline.cpp file this cropped image is passed to next inference through enqueue method 

```
void Pipeline::callback(const std::string& detection_name)
{
name_to_detection_map_
  auto detection_ptr = name_to_detection_map_[detection_name];

  detection_ptr->fetchResults();

  for (auto pos = next_.equal_range(detection_name); pos.first != pos.second;
       ++pos.first)
  {

    std::string next_name = pos.first->second;

    if (output_names_.find(next_name) != output_names_.end())
    {
      detection_ptr->observeOutput(name_to_output_map_[next_name]);
    }
    else
    {
      auto detection_ptr_iter = name_to_detection_map_.find(next_name);

      if (detection_ptr_iter != name_to_detection_map_.end())
      {

        auto next_detection_ptr = detection_ptr_iter->second;

        for (int i = 0; i < detection_ptr->getResultsLength(); ++i)
        {
          const dynamic_vino_lib::Result* prev_result = detection_ptr->getLocationResult(i);
          auto clippedRect = prev_result->getLocation() & cv::Rect(0, 0, width_, height_);
          cv::Mat next_input = frame_(clippedRect);
          next_detection_ptr->enqueue(next_input, prev_result->getLocation());
        }
        
        if (detection_ptr->getResultsLength() > 0)
        {
          increaseInferenceCounter();
          next_detection_ptr->submitRequest();
        }
      }
    }

  }
```
I need to create a for loop in Text Recognition fetchResults() method to pass each of the RoI in Text Recognition network to obtain the output. This for loop will be count using the text detection result quantity. **But How can I access this variable in Text Recognition fetchResults() method?**

In Text Recognition demo example, a main for loop for each rectangle identified in the image was made and the text recognition detection was inside this main loop. So the code cropped the image and pass for text recognition network detection. **How can I do that using ros_openvino toolkit?** 

Thanks in advance.

LewisLiuPub, 2019-11-21:
Do you mean TextDetection produces several ROIs, each of which can be passed to TextRecognition and get one output (a string of characters)? 
If so, before doing more checks, please try to set the "batch" size of TextRecognition a bigger number, for example, 16.

this package has already supported that kind of for loop.
`Pipelines:
- name: text_recognition
  inputs: [StandardCamera]
  infers:
    - name: TextDetection
      model: text_detection_model path
      engine: CPU
      label: to/be/set/xxx.labels
      batch: 1
    - name: TextRecognition
      model: text_recognition_model path
      engine: CPU
      label: to/be/set/xxx.labels
      **batch: 16**
  outputs: [ImageWindow]
  confidence_threshold: 0.5
  connects:
    - left: StandardCamera
      right: [TextDetection] 
    - left: TextDetection
      right: [TextRecognition, ImageWindow]
    - left: TextRecognition
      right: [ImageWindow]

OpenvinoCommon:`

ximenesfel, 2019-11-21:
Hi @LewisLiuPub. Thanks for your comment.

**Do you mean TextDetection produces several ROIs, each of which can be passed to TextRecognition and get one output (a string of characters)?**
Yes. Sorry, I provided images to explain my doubts.

Using the bellow configuration and setting the max_batch_size in TextRecognition constructor to 1.

```
Pipelines:
- name: pipeline name
  inputs: [StandardCamera]
  infers:
    - name: TextDetection
      model: text_detection_model path
      engine: CPU
      label: to/be/set/xxx.labels
      batch: 1
    - name: TextRecognition
      model: text_recognition_model path
      engine: CPU
      label: to/be/set/xxx.labels
      batch: 16
  outputs: [ImageWindow]
  confidence_threshold: 0.5
  connects:
    - left: StandardCamera
      right: [TextDetection] 
    - left: TextDetection
      right: [TextRecognition, ImageWindow]
    - left: TextRecognition
      right: [ImageWindow]

OpenvinoCommon: 
```
When I include only one text in the image, the network inferred correctly.

[image]

When I include more than one text in the image, the network starts to fail. Detect two texts but not include text recognition. 

[image]

Using the bellow configuration and setting the max_batch_size in TextRecognition constructor to 2.

```
Pipelines:
- name: pipeline name
  inputs: [StandardCamera]
  infers:
    - name: TextDetection
      model: text_detection_model path
      engine: CPU
      label: to/be/set/xxx.labels
      batch: 1
    - name: TextRecognition
      model: text_recognition_model path
      engine: CPU
      label: to/be/set/xxx.labels
      batch: 16
  outputs: [ImageWindow]
  confidence_threshold: 0.5
  connects:
    - left: StandardCamera
      right: [TextDetection] 
    - left: TextDetection
      right: [TextRecognition, ImageWindow]
    - left: TextRecognition
      right: [ImageWindow]

OpenvinoCommon: 
```
When I include two texts in the image, the network doesn't infer correctly for both texts. It's the same detection for both texts, so I think the network only receiving one cropped text.

[image]

Looking at the **Text Recognition demo** main program, the overall detection was done in this [for loop](https://github.com/opencv/open_model_zoo/blob/e372d4173e50741a6828cda415d55c37320f89cd/demos/text_detection_demo/main.cpp#L188), where for each rectangle both inferences (Text detection and Text Recognition) are made in sequence.

The problem starts when I try to run the Text Recognition inference for more than one cropped text using the ros_openvino toolkit (For one text the pipeline works fine). The cropped text is enqueued in enqueue() method from TextRecognition object. To obtain the output from the Text recognition network I need to do:

` float *ouput_data_pointer = request->GetBlob(output)->buffer().as<float *>();`

After changing the max_batch_size in Text Recognition constructor my getResultsLength() is correct, the same as text rectangles in images. 

From the output ouput_data_pointer, I received only one float output. Not an output vector that represents each text recognition for each cropped text. So I suppose that I need to pass each cropped text in a for loop and obtain the text recognition inference for each cropped text. **Am I worng?**. 

**How to use ouput_data_pointer for each cropped text in fetchResutls() method to obtain inference correctly for each text ?**

Thanks in advance.

closed 2020-01-17 by ximenesfel