intel/ros_openvino_toolkit #51: Error while optimize mobilenet-ssd

songshan0321, 2019-10-21:
```bash
sudo python3 /opt/openvino_toolkit/dldt/model-optimizer/mo.py --input_model /opt/openvino_toolkit/open_model_zoo/model_downloader/object_detection/common/mobilenet-ssd/caffe/mobilenet-ssd.caffemodel --output_dir /opt/openvino_toolkit/open_model_zoo/model_downloader/object_detection/common/mobilenet-ssd/caffe/output/FP32 --mean_values [127.5,127.5,127.5] --scale_values [127.5]
```
Error output:
```bash
[ ERROR ]  ----------------- INTERNAL ERROR ----------------
[ ERROR ]  Unexpected exception happened.
[ ERROR ]  Please contact Model Optimizer developers and forward the following information:
[ ERROR ]  'MultiDiGraph' object has no attribute 'node'
[ ERROR ]  Traceback (most recent call last):
  File "/home/shan/code/dldt/model-optimizer/mo/main.py", line 325, in main
    return driver(argv)
  File "/home/shan/code/dldt/model-optimizer/mo/main.py", line 279, in driver
    custom_layers_mapping_path=custom_layers_mapping_path)
  File "/home/shan/code/dldt/model-optimizer/mo/pipeline/caffe.py", line 74, in driver
    graph, original_shapes = loader.caffe_pb_to_nx(proto, model)
  File "/home/shan/code/dldt/model-optimizer/mo/front/caffe/loader.py", line 314, in caffe_pb_to_nx
    if len(node.out_nodes()) == 0:
  File "/home/shan/code/dldt/model-optimizer/mo/graph/graph.py", line 313, in out_nodes
    assert self.has('kind')
  File "/home/shan/code/dldt/model-optimizer/mo/graph/graph.py", line 273, in has
    return k in self.graph.node[self.node]
AttributeError: 'MultiDiGraph' object has no attribute 'node'

[ ERROR ]  ---------------- END OF BUG REPORT --------------
```

LewisLiuPub, 2019-10-21:
Thank you very much for submitting this issue.
Actually, we have also found this issue, this issue was caused by the mismatching between the old OpenVINO version and the updated dependent framework/libs.

It can be fixed by upgrading OpenVINO Toolkit version 2019R2 or newer, you can try this way if you are rush for your use, before we make the official fix.

songshan0321, 2019-10-21:
Thanks for your fast response too! Yes I need it quite urgently for my use, may I know what are the steps to upgrade OpenVINO to version 2019R2 safely without breaking other dependencies?

songshan0321, 2019-10-21:
I have upgraded OpenVINO to version 2019R3, however, I still not able to optimize mask_rcnn_inception_v2. 
When I run:
```bash
python3 /opt/intel/openvino_2019.3.334/deployment_tools/model_optimizer/mo_tf.py --input_model frozen_inference_graph.pb --tensorflow_use_custom_operations_config /opt/intel/openvino/deployment_tools/model_optimizer/extensions/front/tf/mask_rcnn_support.json --tensorflow_object_detection_api_pipeline_config ~/Downloads/models/mask_rcnn_inception_v2_coco_2018_01_28/pipeline.config --reverse_input_channels --output_dir ~/Downloads/models/mask_rcnn_inception_v2_coco_2018_01_28/output/
```
I got error:
```bash
Model Optimizer arguments:
Common parameters:
	- Path to the Input Model: 	/home/shan/Downloads/models/mask_rcnn_inception_v2_coco_2018_01_28/frozen_inference_graph.pb
	- Path for generated IR: 	/home/shan/Downloads/models/mask_rcnn_inception_v2_coco_2018_01_28/output/
	- IR output name: 	frozen_inference_graph
	- Log level: 	ERROR
	- Batch: 	Not specified, inherited from the model
	- Input layers: 	Not specified, inherited from the model
	- Output layers: 	Not specified, inherited from the model
	- Input shapes: 	Not specified, inherited from the model
	- Mean values: 	Not specified
	- Scale values: 	Not specified
	- Scale factor: 	Not specified
	- Precision of IR: 	FP32
	- Enable fusing: 	True
	- Enable grouped convolutions fusing: 	True
	- Move mean values to preprocess section: 	False
	- Reverse input channels: 	True
TensorFlow specific parameters:
	- Input model in text protobuf format: 	False
	- Path to model dump for TensorBoard: 	None
	- List of shared libraries with TensorFlow custom layers implementation: 	None
	- Update the configuration file with input/output node names: 	None
	- Use configuration file used to generate the model with Object Detection API: 	/home/shan/Downloads/models/mask_rcnn_inception_v2_coco_2018_01_28/pipeline.config
	- Operations to offload: 	None
	- Patterns to offload: 	None
	- Use the config file: 	/opt/intel/openvino/deployment_tools/model_optimizer/extensions/front/tf/mask_rcnn_support.json
Model Optimizer version: 	2019.3.0-375-g332562022
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:516: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:517: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:518: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:519: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:520: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
/usr/local/lib/python3.5/dist-packages/tensorflow/python/framework/dtypes.py:525: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  np_resource = np.dtype([("resource", np.ubyte, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:541: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint8 = np.dtype([("qint8", np.int8, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:542: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint8 = np.dtype([("quint8", np.uint8, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:543: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint16 = np.dtype([("qint16", np.int16, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:544: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_quint16 = np.dtype([("quint16", np.uint16, 1)])
/usr/local/lib/python3.5/dist-packages/tensorboard/compat/tensorflow_stub/dtypes.py:545: FutureWarning: Passing (type, 1) or '1type' as a synonym of type is deprecated; in a future version of numpy, it will be understood as (type, (1,)) / '(1,)type'.
  _np_qint32 = np.dtype([("qint32", np.int32, 1)])
[... cut ...]
  np_resource = np.dtype([("resource", np.ubyte, 1)])
[ ERROR ]  'Graph' object has no attribute 'node'
Cannot pre-process TensorFlow graph after reading from model file "/home/shan/Downloads/models/mask_rcnn_inception_v2_coco_2018_01_28/frozen_inference_graph.pb". File is corrupt or has unsupported format. Details: 'Graph' object has no attribute 'node'. 
 For more information please refer to Model Optimizer FAQ (https://docs.openvinotoolkit.org/latest/_docs_MO_DG_prepare_model_Model_Optimizer_FAQ.html), question #44.
```

LewisLiuPub, 2019-10-21:
Do you mean, you succeeded in optimizing mobilenet-ssd, but failed for mask_rcnn?

we have not yet try R3, but I am sure R2 is ok for MO, so recommend you to try R2.0.1 version.

LewisLiuPub, 2019-10-21:
also please make sure you source'd the new version's Setupvars.sh after you install R3.

songshan0321, 2019-10-21:
I run the following code to source the R2 version and optimize mobilenet-ssd.
```bash
source /opt/intel/openvino_2019.2.275/bin/setupvars.sh
sudo python3 /opt/openvino_toolkit/dldt/model-optimizer/mo.py --input_model /opt/openvino_toolkit/open_model_zoo/model_downloader/object_detection/common/mobilenet-ssd/caffe/mobilenet-ssd.caffemodel --output_dir /opt/openvino_toolkit/open_model_zoo/model_downloader/object_detection/common/mobilenet-ssd/caffe/output/FP32 --mean_values [127.5,127.5,127.5] --scale_values [127.5]
```
I still got the same error:
```AttributeError: 'MultiDiGraph' object has no attribute 'node'```

LewisLiuPub, 2019-10-22:
Don't use this file /opt/openvino_toolkit/dldt/model-optimizer/mo.py, please use the one under folder /opt/intel/openvino/... instead.

songshan0321, 2019-10-22:
I source the R2 version and optimize mobilenet-ssd by running the script under folder /opt/intel/openvino/...
```bash
source /opt/intel/openvino_2019.2.275/bin/setupvars.sh
sudo python3 /opt/intel/openvino/deployment_tools/model_optimizer/mo_caffe.py --input_model /opt/openvino_toolkit/open_model_zoo/model_downloader/object_detection/common/mobilenet-ssd/caffe/mobilenet-ssd.caffemodel --output_dir /opt/openvino_toolkit/open_model_zoo/model_downloader/object_detection/common/mobilenet-ssd/caffe/output/FP32 --mean_values [127.5,127.5,127.5] --scale_values [127.5]
```
I still got the same error:
```bash
Model Optimizer arguments:
Common parameters:
	- Path to the Input Model: 	/opt/openvino_toolkit/open_model_zoo/model_downloader/object_detection/common/mobilenet-ssd/caffe/mobilenet-ssd.caffemodel
	- Path for generated IR: 	/opt/openvino_toolkit/open_model_zoo/model_downloader/object_detection/common/mobilenet-ssd/caffe/output/FP32
	- IR output name: 	mobilenet-ssd
	- Log level: 	ERROR
	- Batch: 	Not specified, inherited from the model
	- Input layers: 	Not specified, inherited from the model
	- Output layers: 	Not specified, inherited from the model
	- Input shapes: 	Not specified, inherited from the model
	- Mean values: 	[127.5,127.5,127.5]
	- Scale values: 	[127.5]
	- Scale factor: 	Not specified
	- Precision of IR: 	FP32
	- Enable fusing: 	True
	- Enable grouped convolutions fusing: 	True
	- Move mean values to preprocess section: 	False
	- Reverse input channels: 	False
Caffe specific parameters:
	- Path to Python Caffe* parser generated from caffe.proto: 	/opt/intel/openvino/deployment_tools/model_optimizer/mo/front/caffe/proto
	- Enable resnet optimization: 	True
	- Path to the Input prototxt: 	/opt/openvino_toolkit/open_model_zoo/model_downloader/object_detection/common/mobilenet-ssd/caffe/mobilenet-ssd.prototxt
	- Path to CustomLayersMapping.xml: 	Default
	- Path to a mean file: 	Not specified
	- Offsets for a mean file: 	Not specified
Model Optimizer version: 	2019.2.0-436-gf5827d4
[ ERROR ]  -------------------------------------------------
[ ERROR ]  ----------------- INTERNAL ERROR ----------------
[ ERROR ]  Unexpected exception happened.
[ ERROR ]  Please contact Model Optimizer developers and forward the following information:
[ ERROR ]  'Graph' object has no attribute 'node'
[ ERROR ]  Traceback (most recent call last):
  File "/opt/intel/openvino_2019.2.275/deployment_tools/model_optimizer/mo/main.py", line 302, in main
    return driver(argv)
  File "/opt/intel/openvino_2019.2.275/deployment_tools/model_optimizer/mo/main.py", line 261, in driver
    custom_layers_mapping_path=custom_layers_mapping_path)
  File "/opt/intel/openvino_2019.2.275/deployment_tools/model_optimizer/mo/pipeline/caffe.py", line 63, in driver
    graph, original_shapes = loader.caffe_pb_to_nx(proto, model)
  File "/opt/intel/openvino_2019.2.275/deployment_tools/model_optimizer/mo/front/caffe/loader.py", line 252, in caffe_pb_to_nx
    kind='op')
  File "/opt/intel/openvino_2019.2.275/deployment_tools/model_optimizer/mo/graph/graph.py", line 499, in add_node
    node.update_node()
  File "/opt/intel/openvino_2019.2.275/deployment_tools/model_optimizer/mo/graph/graph.py", line 464, in update_node
    in_ports_count = self.in_ports_count if self.has_valid('in_ports_count') else None
  File "/opt/intel/openvino_2019.2.275/deployment_tools/model_optimizer/mo/graph/graph.py", line 155, in has_valid
    return self.has(k) and not self.graph.node[self.node][k] is None
  File "/opt/intel/openvino_2019.2.275/deployment_tools/model_optimizer/mo/graph/graph.py", line 152, in has
    return k in self.graph.node[self.node]
AttributeError: 'Graph' object has no attribute 'node'

[ ERROR ]  ---------------- END OF BUG REPORT --------------
[ ERROR ]  -------------------------------------------------
```

LewisLiuPub, 2019-11-20:
@songshan0321 , OpenVINO R3.1 fixed this bug.
The root cause is the latest versions [version 2.4+) of package networkx can't be supported correctly by OpenVINO toolkit.
You can manually fix it for <R3.1 by manually downgrade nextworkx to any version between[1.11, 2.3].

Currently, this openvino wrapper has already upgraded to support OpenVINO R3.1, so suppose it can work well without any extra actions.

closed 2019-11-20 by LewisLiuPub