intel/ros_openvino_toolkit #52: enable_roi_constraint parameter

ximenesfel, 2019-11-04:
Hi.

I need to use an ROI area from a frame after processing the Object detection model. I looked at ros-openvino code and try to understand where the enable_roi_constraint parameter was used but I'm not able to determine. This parameter was found in these files:

**ros_openvino_toolkit/dynamic_vino_lib/src/inferences/object_detection.cpp**
This is a parameter for Object detection object but it no used in constructor.

**ros_openvino_toolkit/dynamic_vino_lib/src/pipeline_manager.cpp**
This parameter is used in Object Detection method called createObjectDetection

**ros_openvino_toolkit/vino_param_lib/include/vino_param_lib/param_manager.h**

struct InferenceParams
  {
    std::string name;
    std::string engine;
    std::string model;
    std::string label;
    int batch;
    float confidence_threshold = 0.5;
    bool enable_roi_constraint = false;
  };

But Where the Object Detection inference use this information to pass in the pipeline only the ROI area?

Thanks in advance.

LewisLiuPub, 2019-11-05:
Hi, @ximenesfel, 
in some previous versions of openvino toolkit, we found that some ROIs (getting from the detection inference) involve negative values as the start point, for example [-1, -3, 50, 100]. Point(-1, -3) is out of the range of the frame's pixels, which makes the sequential steps confusing. So, as a workaround, we set enable_roi_constraint to true to refine the ROI from [-1, -3, 50, 100] to [0, 0, 50, 100].

If you want to get the subframe only covered by the ROI, I think you can do this way:
1. use this package to do the corresponding inference, and get the ROIs you are interested in.
2. get the original frame which is used for inference in step1.
3. use CV API to get the subframe.
          auto roi = next_rois[i];
          auto clippedRect = roi & cv::Rect(0, 0, width_, height_);
          cv::Mat next_input = frame_(clippedRect);

ximenesfel, 2019-11-05:
Hi @LewisLiuPub. Thanks for your detailed explanation. I'll try to implement your code explanation.

ximenesfel, 2019-11-06:
Hi @LewisLiuPub. Looking at callback() method in pipeline.cpp:

```
void Pipeline::callback(const std::string& detection_name)
{
  auto detection_ptr = name_to_detection_map_[detection_name];

  detection_ptr->fetchResults();

  // set output
  for (auto pos = next_.equal_range(detection_name); pos.first != pos.second;
       ++pos.first)
  {

    std::string next_name = pos.first->second;
    // if next is output, then print
    if (output_names_.find(next_name) != output_names_.end())
    {
      // name_to_output_map_[next_name]->accept(*detection_ptr->getResult());
      detection_ptr->observeOutput(name_to_output_map_[next_name]);
    }
    else
    {
      auto detection_ptr_iter = name_to_detection_map_.find(next_name);

      if (detection_ptr_iter != name_to_detection_map_.end())
      {

        auto next_detection_ptr = detection_ptr_iter->second;

        for (int i = 0; i < detection_ptr->getResultsLength(); ++i)
        {
          const dynamic_vino_lib::Result* prev_result = detection_ptr->getLocationResult(i);
          auto clippedRect = prev_result->getLocation() & cv::Rect(0, 0, width_, height_);
          cv::Mat next_input = frame_(clippedRect);
          next_detection_ptr->enqueue(next_input, prev_result->getLocation());
        }
        
        if (detection_ptr->getResultsLength() > 0)
        {
          increaseInferenceCounter();
          next_detection_ptr->submitRequest();
        }
      }
    }

  }
  decreaseInferenceCounter();
  cv_.notify_all();
}
```
I think the callback method do a similar code that you presented. **Am I wrong?**

The pipeline manager call this method to make the inference for Object detection. I'm creating a new pipeline for openvino using the following structure:

```
Pipelines:
- name: my_model
  inputs: [StandardCamera]
  infers:
    - name: ObjectDetection
      model: path_to_model
      engine: CPU
      label: to/be/set/xxx.labels
      batch: 16
      enable_roi_constraint: true
    - name: TextDetection
      model: /root/models/text_detection/text-detection-0004.xml
      engine: CPU
      label: to/be/set/xxx.labels
      batch: 16
  outputs: [ImageWindow]
  confidence_threshold: 0.5
  connects:
    - left: StandardCamera
      right: [ObjectDetection]
    - left: ObjectDetection
      right: [TextDetection]
    - left: TextDetection
      right: [ImageWindow]

OpenvinoCommon:
```
My objective is to detect an object in an image, obtain this ROI (Object detection bounding box) and pass this ROI to a text detection model (like described above).

I'm using the text recognition demo (https://github.com/opencv/open_model_zoo/tree/master/demos/text_detection_demo) to implement this new pipeline as reference. The object detection is running ok but when I inserted in the pipeline the TextDetection model and try to obtain the text detection rectangles number from detection, I'm receiving zero as text detection rectangles. So, I think the TextDetection model is not receiving the correct ROI information. **How can I verify if TextDetection is receiving the correct information?**

Thanks in advance.

LewisLiuPub, 2019-11-07:
Hi, @ximenesfel 
First you got the point we are designing for, the openvino component can be use that way as you described. Just one suggestion: It would better set batch size to "1" for the first inference ("ObjectDetection" in your case). "batch:16"-->"batch:1"
Second, would you provide some example pictures that you are using? What kind of objects do you want ObjectDetection infer to detect and pass them to TextDetection infer?

Usually, an effective way to debug is to separately test the TextDetection inference's function. After you are sure it works, then you create a pipeline (ObejctDetection-->TextDetection) as the second step.

ximenesfel, 2019-11-07:
Hi, @LewisLiuPub. First, thanks for your comments. Bellow my answers:

**Just one suggestion: It would better set batch size to "1" for the first inference ("ObjectDetection" in your case). "batch:16"-->"batch:1"**
Thanks, I`ve changed this parameter as you suggested.

**Second, would you provide some example pictures that you are using? What kind of objects do you want ObjectDetection infer to detect and pass them to TextDetection infer?**
Sorry. Unfortunately, I can't provide an example picture. But we need to detect a tag and read some text.

**Usually, an effective way to debug is to separately test the TextDetection inference's function. After you are sure it works, then you create a pipeline (ObejctDetection-->TextDetection) as the second step.**
Thanks. I started yesterday to do that. I have the text recognition demo functional in my machine, using only openvino toolkit. So I'm trying to understand where is the problem. One thing got my attention. In text recognition demo, before making the inference, the frame was resized to network input size (1280 x 768).  https://github.com/opencv/open_model_zoo/blob/master/demos/text_detection_demo/src/cnn.cpp (Line 94). I'm using the standardcamera input. Looking at standard_camera.cpp, I've tried to use methods setWidth and setHeight to resize the input frame but the text detection is still not working. **Can I resize the frame input before network inference? If yes, How can I do that?**

Thanks.

ximenesfel, 2019-11-07:
Hi @LewisLiuPub, 

Debugging my implementation, I found that the link_data_pointer variable (https://github.com/opencv/open_model_zoo/blob/master/demos/text_detection_demo/src/text_detection.cpp - line 206) is returning -nan:

`[ INFO ] link_data_pointer: -nan`

I'm using the ros_openvino_toolkit with openvino 2018 R5 version (last release available) and using       engine: CPU , but in text_recognition demo I'm using openvino 2019 R2. When I was configuring the network, I received the following error:

`[ ERROR ] The plugin does not support FP16`

To resolve this error, I changed all precision variable in network .xml file from FP16 to FP32. I declare my input and output as FP32:

```
input_info->setPrecision(InferenceEngine::Precision::FP32);
output_data_ptr->setPrecision(InferenceEngine::Precision::FP32);
```

**Do you think this is the reason why I'm receiving -nan message?**

Thanks in advance.

LewisLiuPub, 2019-11-08:
> Hi @LewisLiuPub,
>
> Debugging my implementation, I found that the link_data_pointer variable (https://github.com/opencv/open_model_zoo/blob/master/demos/text_detection_demo/src/text_detection.cpp - [...]

I don't think it's the reason. setting Precision to FP32 is right for most of models. 

I guess, the working environment of your computer was messy due to more than 1 Openvino version were installed. Make sure your envionment is clearly init'd with the new version of Openvino before you run your application.

LewisLiuPub, 2019-11-08:
> https://github.com/opencv/open_model_zoo/blob/master/demos/text_detection_demo/src/cnn.cpp (Line 94). I'm using the standardcamera input. Looking at standard_camera.cpp, I've tried to use methods [...]

Aha, I guess you found a bug of ros_openvino_toolkit here. Currently, setWidth and setHeight can't correctly set the input frame's width/height, which makes a lot of confusion. the correct way to call this func: **bool Input::StandardCamera::initialize(size_t width, size_t height)**.
Probably, you need to update this line(https://github.com/intel/ros_openvino_toolkit/blob/master/dynamic_vino_lib/src/pipeline_manager.cpp#L125), or the like in your own code:
replace device->initialize() to device->initialize(1280,768).

ximenesfel, 2019-11-08:
> I don't think it's the reason. setting Precision to FP32 is right for most of models.
>
> I guess, the working environment of your computer was messy due to more than 1 Openvino version were installed. [...]

Sorry, I forgot to explain my machine configuration. I using two separated docker containers (one for ros_openvino  and other to text_recognition_demo). But looking at the information provided by  ros_openvino compilation: 

```
InferenceEngine: 
	API version ............ 1.4
	Build .................. custom_HEAD_30594bb3099b7874f372074e6c0cc82748344435
[ INFO ] Pipeline size: 1
[ INFO ] Inferences size: 1
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ INFO ] Inference Params:name=name
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ INFO ] Pipeline Params:name=tname
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ INFO ] --------parameters DUMP---------------------
[ INFO ] Pipeline: pipeline_name
[ INFO ] 	Inputs: StandardCamera, 
[ INFO ] 	Outputs: ImageWindow, 
[ INFO ] 	Inferences: 
[ INFO ] 		Name: TextDetection
[ INFO ] 		Model: path to text-detection-0004.xml
[ INFO ] 		Engine: CPU
[ INFO ] 		Label: to/be/set/xxx.labels
[ INFO ] 		Batch: 1
[ INFO ] 		Confidence_threshold: 0.5
[ INFO ] 		Enable_roi_constraint: 0
[ INFO ] 	Connections: 
[ INFO ] 		StandardCamera->TextDetection
[ INFO ] 		TextDetection->ImageWindow
[ INFO ] Common:
[ INFO ] 	camera_topic: 
[ INFO ] 	custom_cpu_library: 
[ INFO ] 	custom_cldnn_library: 
[ INFO ] 	enable_performance_count: 0
[ INFO ] Parsing InputDvice: StandardCamera
[ INFO ]  ... Adding one Input device: StandardCamera
[ INFO ] Parsing Output: ImageWindow
[ INFO ]  ... Adding one Output: ImageWindow
[ INFO ] Parsing Inference: TextDetection

	API version ............ 1.5
	Build .................. custom_HEAD_30594bb3099b7874f372074e6c0cc82748344435
	Description ....... MKLDNNPlugin
[ INFO ] Test 
[ INFO ] Loading network files
[ INFO ] Batch size is set to  1
[ INFO ] Checking input size
[ INFO ] Checking output size
[ INFO ] Checking Text Detection outputs
[ INFO ] Checking Text Detection outputs ...
[ INFO ] Checking Text Detection output ... Name=model/link_logits_/add
[ INFO ]  ... Adding one Inference: TextDetection
[ INFO ] Updating connections ...
[ INFO ] Checking connection into pipeline:[StandardCamera(1)<-->TextDetection(2)]
[ INFO ] Adding connection into pipeline:[StandardCamera<-->TextDetection]
[ INFO ] Checking connection into pipeline:[TextDetection(2)<-->ImageWindow(3)]
[ INFO ] Adding connection into pipeline:[TextDetection<-->ImageWindow]
[ INFO ] One Pipeline Created!
 --> StandardCamera
StandardCamera --> TextDetection
TextDetection --> ImageWindow
```
When the code parse Text Detection model the code show that the API version is 1.5. The configured version in text-detection-0004.xml tis 4:

`<net batch="1" name="text-detection-0004" version="4">`

In this configuration, I'm using openvino dldt branch 2018_R5 and ros_openvino_toolkit Version 0.5.0_R5.

**Do you think this could be the problem? How can I change this API version to 1.4 to match with inference Engine?**

> Aha, I guess you found a bug of ros_openvino_toolkit here. Currently, setWidth and setHeight can't correctly set the input frame's width/height, which makes a lot of confusion. the correct way to [...]

Glad to know that I helped to find a bug in ros_openvino_package. I've changed the intialize() method as you suggested and the input changed but the -nan problem persists.

songshan0321, 2019-11-19:
I have written a ROS wrapper for OpenVINO. It is pretty straight forward to use.
https://github.com/songshan0321/ros_vino

LewisLiuPub, 2019-11-20:
@ximenesfel Sorry for late reply to you. 
We have spent a few days in upgrading this package to support OpenVINO R3.1. It would be a good way for you to double check your issues with the latest version. I'll be happy if you share your further results.

BTW, please use branch: **[devel](https://github.com/intel/ros_openvino_toolkit/tree/devel)**:

LewisLiuPub, 2019-11-20:
@songshan0321 
Very glad to hear you created another ROS wrapper for OpenVINO. ^,^.
I hope more and more people involve this kind of attempts and development, then we have more opportunities to make ROS+AI run better.

ximenesfel, 2019-11-20:
@LewisLiuPub Thanks for your update. So I upgrade the openvino for version 2019.R2 and worked. Glad to know that the ros_openvino package was upgraded to support  OpenVINO R3.1.

@songshan0321 Thanks for sharing your wrapper. Great work.

closed 2019-11-20 by ximenesfel