intel/ros_openvino_toolkit #58: Unable to build Inference Engine

soyjorge, 2020-01-06:
While building OpenVINO's DLDT, after entering the command

cmake -DCMAKE_BUILD_TYPE=Release ..

I'm getting the following error:
-- Configuring incomplete, errors occurred!
See also "/home/jorge/code/dldt/inference-engine/build/CMakeFiles/CMakeOutput.log".
See also "/home/jorge/code/dldt/inference-engine/build/CMakeFiles/CMakeError.log".

Is there another way to build the package?

LewisLiuPub, 2020-01-07:
@soyjorge did you try the build process by following [DLDT's Readme](https://github.com/opencv/dldt/blob/2019/inference-engine/README.md)?

sradmard, 2020-02-13:
@soyjorge the problem is that the DLDT package relies on some submodules that you need to initialise.  Then you will be able to build it.
```
cd dldt/inference-engine/
git submodule update --init --recursive
```

LewisLiuPub, 2020-02-20:
Close this issue, please feel free to re-open it if you have more comments.

closed 2020-02-20 by LewisLiuPub