intel/ros_openvino_toolkit #59: Compiling the Package to Use NCS2 (MYRIAD).

sradmard, 2020-01-10:
My system: 
Ubuntu 16.04
ROS Kinetic

Hi,
I am having difficulty on running the package with NCS2 (MYRIAD configuration). There are two relevant issues that have been closed, but going through their instructions also does not help. Here is the situation:
- If I follow the instruction from [here](https://github.com/intel/ros_openvino_toolkit#enable-intel-neural-compute-stick-2-intel-ncs-2-under-the-openvino-open-source-version-optional-), and install dependencies from [here](https://github.com/intel/ros_openvino_toolkit/blob/master/doc/OPEN_SOURCE_CODE_README.md) as instructed (this follows the OPEN_SOURCE_CODE_README.md file), then I can catkin_make the ros_openvino_toolkit successfully. The problem is that it does not recognize the MYRIAD configuration as also mentioned in this [issue](https://github.com/intel/ros_openvino_toolkit/issues/44). The solution provided in that issue is to install the binary version, which I think is following the instructions [here.](https://github.com/intel/ros_openvino_toolkit/blob/master/doc/BINARY_VERSION_README.md). 

When I follow the Binary installation, I get the following error during catkin_make of the ros_openvino_toolkit. 
`In file included from /home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/src/pipeline_with_params.cpp:60:0:
/home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/include/sample/utility.hpp:159:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_uint32(n_ag, 16, num_batch_ag_message);
              ^
/home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/include/sample/utility.hpp:162:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_uint32(n_hp, 16, num_batch_hp_message);
              ^
/home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/include/sample/utility.hpp:165:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_uint32(n_em, 16, num_batch_em_message);
              ^
In file included from /home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/src/image_reidentification_server.cpp:60:0:
/home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/include/sample/utility.hpp:159:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_uint32(n_ag, 16, num_batch_ag_message);
              ^
/home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/include/sample/utility.hpp:162:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_uint32(n_hp, 16, num_batch_hp_message);
              ^
/home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/include/sample/utility.hpp:165:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_uint32(n_em, 16, num_batch_em_message);
              ^
In file included from /home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/src/image_people_server.cpp:60:0:
/home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/include/sample/utility.hpp:159:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_uint32(n_ag, 16, num_batch_ag_message);
              ^
/home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/include/sample/utility.hpp:162:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_uint32(n_hp, 16, num_batch_hp_message);
              ^
/home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/include/sample/utility.hpp:165:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_uint32(n_em, 16, num_batch_em_message);
              ^
ros_openvino_toolkit/sample/CMakeFiles/pipeline_with_params.dir/build.make:62: recipe for target 'ros_openvino_toolkit/sample/CMakeFiles/pipeline_with_params.dir/src/pipeline_with_params.cpp.o' failed
make[2]: *** [ros_openvino_toolkit/sample/CMakeFiles/pipeline_with_params.dir/src/pipeline_with_params.cpp.o] Error 1
CMakeFiles/Makefile2:7700: recipe for target 'ros_openvino_toolkit/sample/CMakeFiles/pipeline_with_params.dir/all' failed
make[1]: *** [ros_openvino_toolkit/sample/CMakeFiles/pipeline_with_params.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
ros_openvino_toolkit/sample/CMakeFiles/image_reidentification_server.dir/build.make:62: recipe for target 'ros_openvino_toolkit/sample/CMakeFiles/image_reidentification_server.dir/src/image_reidentification_server.cpp.o' failed
make[2]: *** [ros_openvino_toolkit/sample/CMakeFiles/image_reidentification_server.dir/src/image_reidentification_server.cpp.o] Error 1
CMakeFiles/Makefile2:7626: recipe for target 'ros_openvino_toolkit/sample/CMakeFiles/image_reidentification_server.dir/all' failed
make[1]: *** [ros_openvino_toolkit/sample/CMakeFiles/image_reidentification_server.dir/all] Error 2
ros_openvino_toolkit/sample/CMakeFiles/image_people_server.dir/build.make:62: recipe for target 'ros_openvino_toolkit/sample/CMakeFiles/image_people_server.dir/src/image_people_server.cpp.o' failed
make[2]: *** [ros_openvino_toolkit/sample/CMakeFiles/image_people_server.dir/src/image_people_server.cpp.o] Error 1
CMakeFiles/Makefile2:7774: recipe for target 'ros_openvino_toolkit/sample/CMakeFiles/image_people_server.dir/all' failed
make[1]: *** [ros_openvino_toolkit/sample/CMakeFiles/image_people_server.dir/all] Error 2
In file included from /home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/src/image_segmentation_server.cpp:60:0:
/home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/include/sample/utility.hpp:159:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_uint32(n_ag, 16, num_batch_ag_message);
              ^
In file included from /home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/src/image_object_server.cpp:60:0:
/home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/include/sample/utility.hpp:159:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_uint32(n_ag, 16, num_batch_ag_message);
              ^
/home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/include/sample/utility.hpp:162:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_uint32(n_hp, 16, num_batch_hp_message);
              ^
/home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/include/sample/utility.hpp:162:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_uint32(n_hp, 16, num_batch_hp_message);
              ^
/home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/include/sample/utility.hpp:165:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_uint32(n_em, 16, num_batch_em_message);
              ^
/home/admin3srp/test_catkin_ws/src/ros_openvino_toolkit/sample/include/sample/utility.hpp:165:14: error: expected constructor, destructor, or type conversion before ‘(’ token
 DEFINE_uint32(n_em, 16, num_batch_em_message);
              ^
ros_openvino_toolkit/sample/CMakeFiles/image_object_server.dir/build.make:62: recipe for target 'ros_openvino_toolkit/sample/CMakeFiles/image_object_server.dir/src/image_object_server.cpp.o' failed
make[2]: *** [ros_openvino_toolkit/sample/CMakeFiles/image_object_server.dir/src/image_object_server.cpp.o] Error 1
CMakeFiles/Makefile2:7372: recipe for target 'ros_openvino_toolkit/sample/CMakeFiles/image_object_server.dir/all' failed
make[1]: *** [ros_openvino_toolkit/sample/CMakeFiles/image_object_server.dir/all] Error 2
ros_openvino_toolkit/sample/CMakeFiles/image_segmentation_server.dir/build.make:62: recipe for target 'ros_openvino_toolkit/sample/CMakeFiles/image_segmentation_server.dir/src/image_segmentation_server.cpp.o' failed
make[2]: *** [ros_openvino_toolkit/sample/CMakeFiles/image_segmentation_server.dir/src/image_segmentation_server.cpp.o] Error 1
CMakeFiles/Makefile2:7987: recipe for target 'ros_openvino_toolkit/sample/CMakeFiles/image_segmentation_server.dir/all' failed
make[1]: *** [ros_openvino_toolkit/sample/CMakeFiles/image_segmentation_server.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
`

In the Binary installation it is asking for an even older version of openvino "2018 R4" which is not available anymore, and it is inconsistent with the content of `environment_setup_binary.sh` script were it downloads "2018 R5".

I am wondering whether anybody has been able to install this package and use it with NCS2 recently or not. 

Thank you in advance.

LewisLiuPub, 2020-02-04:
Thank you very much for your helpful information.
Actually, per our team rule, we mainly put efforts on ROS2 support, and just put limit effort and pay partial attention to ROS version porting.

Recently, we have upgraded the ROS package to support latest OpenVINO toolkit, suppose the build errors would be fixed.

sradmard, 2020-02-05:
Thank you so much for your reply. 
I am going through the ros_openvino_toolkit package, realizing that still in the description as well as in the `script/environment_setup_binary.sh` it is referring to the "l_openvino_toolkit_p_2018.5.455.tgz" version of openvino. 
Regarding dldt, the current package is at 2019_R3.1 version, while the instructions on ros_openvino_toolkit is referring to the 2018_R5 version. 
Also, open_model_zoo package is now at its 2019_R3.1 version, while the instructions are referring to 2018_R5 version. There is a restructure of open_model_zoo package after 2019_R1 version, where they no longer keep `downloader.py` under `model_downloader` directory. They have moved it under `tools/downloader` directory. However, the instruction on the description of ros_openvino_toolkit is still referring to the old version. 
Is it only the instructions that are still not updated? 
Thank you.

LewisLiuPub, 2020-02-07:
@sradmard Yes, we verified the code with its functionalities, but missing to update the description due to time limit and some changes.
Appreciate if some guys here have interests to contribute the missing parts. Otherwise, I will arrange some effort to finish them _after_ some days when we are ready for this task.

Thanks.

sradmard, 2020-02-13:
I am finally able to install a more recent version (2019_R3.1) only through source installation and it is working with MYRIAD as well. I did not need to install OpenVINO though. Here are my steps on Ubuntu 16.04 following a revised version of [this file](https://github.com/intel/ros_openvino_toolkit/blob/master/doc/OPEN_SOURCE_CODE_README.md): 
1- Install ROS Kinetic
2- Install OpenCV (latest version)
3- Install OpenCL
4- Install DLDT (version 2019_R3.1)
5- Install Open Model Zoo (version 2019_R3.1)
6- Install Intel® RealSense™ SDK 2.0 (Latest stable version)
7- Installed [Other Dependencies](https://github.com/intel/ros_openvino_toolkit/blob/master/doc/OPEN_SOURCE_CODE_README.md)
8- Set the environment in my .bashrc
9- Install ROS_OpenVINO packages
10- [Configure the Neural Compute Stick USB Driver](https://github.com/intel/ros_openvino_toolkit)
11- Updated the [preparation steps](https://github.com/intel/ros_openvino_toolkit) and yaml files to direct to the correct .xml file paths since Open Model Zoo has restructured itself after version 2019_R1. 

I can make a pull request, with the updated instructions and yaml files, however, I am still confused since I did not need to install OpenVINO toolkit package. 
I am also not still able to install through the [Tarball (Binary) installation](https://github.com/intel/ros_openvino_toolkit/blob/master/doc/BINARY_VERSION_README.md) with OpenVINO toolkit only. When following the binary instruction, I still get the error in my first post while compiling ros_openvino_toolkit.

LewisLiuPub, 2020-02-20:
yes, you only need to choose one of the options (Binary verison or DLDT version) to install OpenVINO toolkit. If you have install DLDT from source code, then it's unnecessary to install binary version.

Thank you very much for your time and effort.