intel/ros_openvino_toolkit #62: i7 + Movidius ncs2 = Object Detection Model -> cpu 398%

snmsung716, 2020-02-18:
In the example, which is "roslaunch vino_launch pipeline_object_oss.launch", while running this roslaunch file, Ive had 398% of CPU capacity having with the use of movidius ncs 2. Also, for an example, Face Detection " roslaunch vino_launch pipeline_image_oss.launch", Ive gotten 938% of CPU capacity with ncs2. Just I wonder whether Ive made use of movidius correctly or not. while running both of roslaunch files,  vino_launch pipeline_object_oss.launch and vino_launch pipeline_image_oss.launch, Ive had the following warning (invalid node). 

Best regards.

SUMMARY
========

PARAMETERS
 * /pipeline_with_params/param_file: /home/syscon/catk...
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /rviz/default_rviz: /home/syscon/catk...

NODES
  /
    pipeline_with_params (dynamic_vino_sample/pipeline_with_params)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [13648]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 94e65de2-521f-11ea-8c61-208756f4ebbb
process[rosout-1]: started with pid [13733]
started core service [/rosout]
process[pipeline_with_params-2]: started with pid [13736]
process[rviz-3]: started with pid [13753]
[ INFO ] FLAGS_config=/home/syscon/catkin_ws/src/ros_openvino_toolkit/vino_launch/param/pipeline_object_oss.yaml
InferenceEngine: 
	API version ............ 1.4
	Build .................. custom_HEAD_91160e65cf66e7fbc709245a8ba948a4c555e443
[ INFO ] Pipeline size: 1
[ INFO ] Inferences size: 1
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ INFO ] Inference Params:name=ObjectDetection
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ INFO ] Pipeline Params:name=object
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ INFO ] --------parameters DUMP---------------------
[ INFO ] Pipeline: object
[ INFO ] 	Inputs: RealSenseCamera, 
[ INFO ] 	Outputs: ImageWindow, RosTopic, RViz, 
[ INFO ] 	Inferences: 
[ INFO ] 		Name: ObjectDetection
[ INFO ] 		Model: /opt/openvino_toolkit/open_model_zoo/model_downloader/object_detection/common/mobilenet-ssd/caffe/output/FP32/mobilenet-ssd.xml
[ INFO ] 		Engine: CPU
[ INFO ] 		Label: to/be/set/xxx.labels
[ INFO ] 		Batch: 16
[ INFO ] 		Confidence_threshold: 0.5
[ INFO ] 		Enable_roi_constraint: 0
[ INFO ] 	Connections: 
[ INFO ] 		ObjectDetection->ImageWindow
[ INFO ] 		ObjectDetection->RosTopic
[ INFO ] 		ObjectDetection->RViz
[ INFO ] 		RealSenseCamera->ObjectDetection
[ INFO ] Common:
[ INFO ] 	camera_topic: 
[ INFO ] 	custom_cpu_library: 
[ INFO ] 	custom_cldnn_library: 
[ INFO ] 	enable_performance_count: 0
[ INFO ] Parsing InputDvice: RealSenseCamera
[ INFO] [1582010619.949285810]: rviz version 1.12.17
[ INFO] [1582010619.949331809]: compiled against Qt version 5.5.1
[ INFO] [1582010619.949344388]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1582010620.493798231]: Stereo is NOT SUPPORTED
[ INFO] [1582010620.493948064]: OpenGl version: 3 (GLSL 1.3).
[ INFO ]  ... Adding one Input device: RealSenseCamera
[ INFO ] Parsing Output: ImageWindow
[ INFO ]  ... Adding one Output: ImageWindow
[ INFO ] Parsing Output: RosTopic
[ INFO ]  ... Adding one Output: RosTopic
[ INFO ] Parsing Output: RViz
[ INFO ]  ... Adding one Output: RViz
[ INFO ] Parsing Inference: ObjectDetection

	API version ............ 1.4
	Build .................. custom_HEAD_91160e65cf66e7fbc709245a8ba948a4c555e443
	Description ....... MKLDNNPlugin
[ INFO ] Loading network files
[ INFO ] Batch size is set to  1
[ INFO ] Checking input size
[ INFO ] Checking output size
[ INFO ] Checking Object Detection outputs
[ INFO ] Checking Object Detection outputs ...
[ INFO ] Checking Object Detection output ... Name=detection_out
[ INFO ] Checking Object Detection num_classes
[ INFO ] Checking Object Detection output ... num_classes=21
[ INFO ] max proposal count is: 100
[ INFO ]  ... Adding one Inference: ObjectDetection
[ INFO ] Updating connections ...
[ INFO ] Checking connection into pipeline:[ObjectDetection(2)<-->ImageWindow(3)]
[ INFO ] Adding connection into pipeline:[ObjectDetection<-->ImageWindow]
[ INFO ] Checking connection into pipeline:[ObjectDetection(2)<-->RosTopic(3)]
[ INFO ] Adding connection into pipeline:[ObjectDetection<-->RosTopic]
[ INFO ] Checking connection into pipeline:[ObjectDetection(2)<-->RViz(3)]
[ INFO ] Adding connection into pipeline:[ObjectDetection<-->RViz]
[ INFO ] Checking connection into pipeline:[RealSenseCamera(1)<-->ObjectDetection(2)]
[ INFO ] Adding connection into pipeline:[RealSenseCamera<-->ObjectDetection]
[ INFO ] One Pipeline Created!
 --> RealSenseCamera

snmsung716, 2020-02-19:
I guess this ROS toolkit is not compatible with ncs2 because on the face detection, it has just 4 FPS in average.(https://github.com/intel/ros_openvino_toolkit/blob/master/data/images/object_segmentation.git) while using a Face Detection Launch File on CPU without ncs2, I have 3 FPS either.

LewisLiuPub, 2020-02-20:
@snmsung716 
You need to make sure "MYRIAD" engine is used in the correspondent param file.
By default, "CPU" engine is used for object detection[param file](https://github.com/intel/ros_openvino_toolkit/blob/master/vino_launch/param/pipeline_object_oss.yaml).
update [line 7](https://github.com/intel/ros_openvino_toolkit/blob/master/vino_launch/param/pipeline_object_oss.yaml#L7) from:
`engine: CPU`
to:
`engine: MYRIAD`

snmsung716, 2020-02-24:
I appreciate your quick reply! 
I tried to convert engine : CPU to MYRIAD, but unfortunately Ive had below error. So I thought I set up wrongly and have reinstalled movidius with [the following instruction](https://github.com/intel/ros_openvino_toolkit/blob/master/doc/OPEN_SOURCE_CODE_README.md), and [./environment_setup_binary.sh](https://github.com/intel/ros_openvino_toolkit/blob/master/doc/OPEN_SOURCE_CODE_README.md) instead of environment_setup.sh. I have same error below though.  

started roslaunch server http://scorpion:41681/

SUMMARY
========

PARAMETERS
 * /pipeline_with_params/param_file: /home/syscon/catk...
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /rviz/default_rviz: /home/syscon/catk...

NODES
  /
    pipeline_with_params (dynamic_vino_sample/pipeline_with_params)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [7637]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 8cfaa412-56d4-11ea-9d58-40a5efdb8c43
process[rosout-1]: started with pid [7666]
started core service [/rosout]
process[pipeline_with_params-2]: started with pid [7677]
process[rviz-3]: started with pid [7685]
[ INFO ] FLAGS_config=/home/syscon/catkin_ws/src/ros_openvino_toolkit/vino_launch/param/pipeline_object_oss.yaml
InferenceEngine: 
	API version ............ 1.4
	Build .................. custom_HEAD_91160e65cf66e7fbc709245a8ba948a4c555e443
[ INFO ] Pipeline size: 1
[ INFO ] Inferences size: 1
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ INFO ] Inference Params:name=ObjectDetection
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ INFO ] Pipeline Params:name=object
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ WARNING ] invalid node; this may result from using a map iterator as a sequence iterator, or vice-versa
[ INFO ] --------parameters DUMP---------------------
[ INFO ] Pipeline: object
[ INFO ] 	Inputs: RealSenseCamera, 
[ INFO ] 	Outputs: ImageWindow, RosTopic, RViz, 
[ INFO ] 	Inferences: 
[ INFO ] 		Name: ObjectDetection
[ INFO ] 		Model: /opt/openvino_toolkit/open_model_zoo/model_downloader/object_detection/common/mobilenet-ssd/caffe/output/FP32/mobilenet-ssd.xml
[ INFO ] 		Engine: MYRIAD
[ INFO ] 		Label: to/be/set/xxx.labels
[ INFO ] 		Batch: 16
[ INFO ] 		Confidence_threshold: 0.5
[ INFO ] 		Enable_roi_constraint: 0
[ INFO ] 	Connections: 
[ INFO ] 		ObjectDetection->ImageWindow
[ INFO ] 		ObjectDetection->RosTopic
[ INFO ] 		ObjectDetection->RViz
[ INFO ] 		RealSenseCamera->ObjectDetection
[ INFO ] Common:
[ INFO ] 	camera_topic: 
[ INFO ] 	custom_cpu_library: 
[ INFO ] 	custom_cldnn_library: 
[ INFO ] 	enable_performance_count: 0
[ INFO ] Parsing InputDvice: RealSenseCamera
[ INFO] [1582528150.127566797]: rviz version 1.12.17
[ INFO] [1582528150.127631332]: compiled against Qt version 5.5.1
[ INFO] [1582528150.127657439]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1582528150.474740156]: Stereo is NOT SUPPORTED
[ INFO] [1582528150.474853783]: OpenGl version: 3 (GLSL 1.3).
[ INFO ]  ... Adding one Input device: RealSenseCamera
[ INFO ] Parsing Output: ImageWindow
[ INFO ]  ... Adding one Output: ImageWindow
[ INFO ] Parsing Output: RosTopic
[ INFO ]  ... Adding one Output: RosTopic
[ INFO ] Parsing Output: RViz
[ INFO ]  ... Adding one Output: RViz
[ INFO ] Parsing Inference: ObjectDetection
[ ERROR ] Cannot find plugin to use :

/home/syscon/code/dldt/inference-engine/include/ie_plugin_dispatcher.hpp:86
[pipeline_with_params-2] process has died [pid 7677, exit code 1, cmd /home/syscon/catkin_ws/devel/lib/dynamic_vino_sample/pipeline_with_params /openvino_toolkit/detected_objects:=/ros_openvino_toolkit/detected_objects /openvino_toolkit/images:=/ros_openvino_toolkit/image_rviz __name:=pipeline_with_params __log:=/home/syscon/.ros/log/8cfaa412-56d4-11ea-9d58-40a5efdb8c43/pipeline_with_params-2.log].
log file: /home/syscon/.ros/log/8cfaa412-56d4-11ea-9d58-40a5efdb8c43/pipeline_with_params-2*.log

snmsung716, 2020-02-26:
@LewisLiuPub 
I appreciate that. It works well!