intel/ros_openvino_toolkit #63: Multiple packages found with the same name "people_msgs"

sradmard, 2020-02-19:
Hi, 
Given that there is already a ros package called [people_msgs](http://wiki.ros.org/people_msgs), and some developers like me may be using that in their ros environment beside your package, it will create a name conflict, causing ros returning the following error during catkin_make:
`Multiple packages found with the same name "people_msgs"`

I was wondering whether it is possible to change your people_msgs package name to something more specific, like vino_people_msgs in the specific package.xml and CMakeLists.txt files?

Thank you in advance.

sradmard, 2020-02-19:
I have made a [pull request](https://github.com/intel/ros_openvino_toolkit/pull/64) after changing the name and all references to `vino_people_msgs`.

LewisLiuPub, 2020-02-20:
Thank you very much for your contribution. @sradmard

LewisLiuPub, 2020-02-20:
Actually, I have been thinking what is the best way to manage these messages. I have ever intended to merge people_msgs into [object_msgs](https://github.com/intel/object_msgs), which sounds like providing more separation from the interface and the implementation.

For the short term, we may merge your PR to resolve the naming conflicts. We can also keep it open for long term solution.
Let me know your comments if you have any.

sradmard, 2020-02-20:
The problem with merging people_msgs type to object_msgs is that they are different family of msgs. Intel object_msgs package, seems to be a stable msgs package, and I'm not sure which other projects rely on it. On the other hand, people_msgs in this package, seems to be dedicated to the vino deep learning models. It might be better to stay with it for now. 
I just followed the `vino_launch` package example that you have, and renamed `people_msgs` to `vino_people_msgs`. (I did not renamed the name of the directory, similar to your `sample` package)
The only thing bothering is the inconsistency of `vino_` versus `dynamic_vino_`, for example `vino_param_lib` versus `dynamic_vino_sample` package name. If you want I can take care of the whole renaming. Just let me know which ones need to be renamed to what. 
Cheers

LewisLiuPub, 2020-02-21:
Yes, you are right.
It's important to make the names sync'd.
I do think vino_ is better, dynamic_vino_ is a bit long, and the word "dynamic" would be confusing to people.
so here is my option: 

- [ ] dynamic_vino_lib -->vino_core_lib
- [ ] sample --> vino_sample
- [ ] people_msgs --> vino_people_msgs

others keep unchanged.
do you think it feasible?

sradmard, 2020-02-24:
I like the idea. How about the directory names in the root directory? Should those change as well?

LewisLiuPub, 2020-02-25:
I think so. ^,^.