intel/ros_openvino_toolkit #65: Getting Close to the Edges in the Image, Returns Wrong Values.

sradmard, 2020-02-21:
Hi,
The topics that return ROI, for example the `/ros_openvino_toolkit/face_detection` return wrong values when the detected face gets close to the zero X and Y axis. The `object_vector.roi.x_offset` and `object_vector.roi.y_offset` return upper limit of unsigned int (4294967295), when it should be close to 0. 
The problem is that cv::Rect is an unsigned integer, but it gets assigned negative integers sometimes when the face gets close to zero X, or zero Y. As a result, it publishes numbers close to 4294967295 as the x_offset and y_offset. 
I believe the problem for face detection application is in the line #98 & #99 of the [face_detection.cpp](https://github.com/intel/ros_openvino_toolkit/blob/master/dynamic_vino_lib/src/inferences/face_detection.cpp) code. `r` is declared as `cv::Rect` which is unsigned int, while `static_cast<int>` can return negative int values. 
I was able to fix it in one of the following two ways:
1- 
```
    r.x = abs(static_cast<int>(detections[i * object_size_ + 3] * width_));
    r.y = abs(static_cast<int>(detections[i * object_size_ + 4] * height_)); 
```
or 2-
```
    int x_ = static_cast<int>(detections[i * object_size_ + 3] * width_);
    int y_ = static_cast<int>(detections[i * object_size_ + 4] * height_);
    r.x = (x_ & -(0 < x_));
    r.y = (y_ & -(0 < y_));
```
This is the same with any other application like object_detection where the ROI is calculated and published. 
Any suggestion on the best way to fix this?

LewisLiuPub, 2020-02-21:
@sradmard I think the 2nd way is more meaningful. if the start point of ROI is negative, then 0 is better than make it abs().

LewisLiuPub, 2022-01-25:
Thank you very much for your contribution and sorry for late reply.
by upgrading source code to support new ROS versions and OpenVINO versions, suppose this bug has already been fixed. I'll close this bug and the related PR, if we (you and everyone here) meet it again, then we can reopen it then find better solutions.

closed 2022-01-25 by LewisLiuPub