intel/ros_openvino_toolkit #66: pipeline_with_params immediately exits

impanel, 2020-02-26:
I'm trying to run the `pipeline_people_oss.launch` demo. The `pipeline_with_params` node almost immediately exits however.

Here is the roslaunch log:
[roslaunch.log](https://github.com/intel/ros_openvino_toolkit/files/4258758/roslaunch.log)

Here is `pipeline_people_oss.yaml`:
[pipeline_people_oss.txt](https://github.com/intel/ros_openvino_toolkit/files/4258766/pipeline_people_oss.txt)

I see in the documentation that the label attributes are unused. Other than that I cannot see anything odd about the param file.

Thanks in advance

sradmard, 2020-02-27:
This is probably because the path to your models are not correct and therefor the models are not available. 
Depending on which version of OpenVINO or open_model_zoo you have installed, the path could have changed. Look for the following files in your system and adjust their path in your yaml file.
`face-detection-adas-0001.xml`
`age-gender-recognition-retail-0013.xml`
`emotions-recognition-retail-0003.xml`
`head-pose-estimation-adas-0001.xml`

impanel, 2020-02-28:
Thanks for your reply, but I'm pretty sure the models are loaded correctly. 

I've tried other examples (e.g. pipeline_image.launch, pipeline_video.launch) and RViz starts without errors but none of the OpenVINO topics are listed. 

It also appears the pipeline exits immediately and the console prints `[ INFO ] Execution successful` without ever displaying the inferred results on the image or video.

Any ideas what might be wrong?
Thanks!

See the log here:
[ros_log_vino_image.txt](https://github.com/intel/ros_openvino_toolkit/files/4264909/ros_log_vino_image.txt)

System: 
- I'm running all this from within a docker container
- Ubuntu 18.04
- openVINO SRB5.0
- dldt 2018_R5
- open_model_zoo 2018_R5

impanel, 2020-03-03:
I'm thinking the issue might be somehow related to the docker container. Here is my Docker file: 
[Dockerfile.txt](https://github.com/intel/ros_openvino_toolkit/files/4283879/Dockerfile.txt)

Does anyone see anything wrong, or has anyone experienced issues with OpenVINO and docker?

Thanks

Eric97, 2020-03-04:
Encountered the same problem as well. But I am using 2018_R4 without docker.

andras-kondakor, 2020-08-31:
Any solution to this problem? I just run into this.