intel/ros_openvino_toolkit #75: catkin_make error with ros_openvino

HermanLiu1, 2020-05-21:
I am trying to build the package. I have followed the guide but unfortunately I encounter a build error return-statement with no value, in function returning ‘void*’ [-fpermissive].I am using Tarball openVINO R3.1 2019 and the master branch from this project.The pipeline_object.launch can start normally.

> ####
#### Running command: "make cmake_check_build_system" in "/home/ysxt001/openvino_ws/build"
####
####
#### Running command: "make -j12 -l12" in "/home/ysxt001/openvino_ws/build"
####
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_py
[  0%] Built target sensor_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target sensor_msgs_generate_messages_eus
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target tf_generate_messages_nodejs
[  0%] Built target diagnostic_msgs_generate_messages_eus
[  0%] Built target _catkin_empty_exported_target
[  0%] Built target diagnostic_msgs_generate_messages_nodejs
[  0%] Built target dynamic_reconfigure_gencfg
[  0%] Built target dynamic_reconfigure_generate_messages_py
[  0%] Built target _object_msgs_generate_messages_check_deps_DetectObject
[  0%] Built target dynamic_reconfigure_generate_messages_lisp
[  0%] Built target _object_msgs_generate_messages_check_deps_ObjectsInBoxes
[  0%] Built target _object_msgs_generate_messages_check_deps_Object
[  0%] Built target dynamic_reconfigure_generate_messages_nodejs
[  0%] Built target _object_msgs_generate_messages_check_deps_Objects
[  0%] Built target _object_msgs_generate_messages_check_deps_ObjectInBox
[  0%] Built target _object_msgs_generate_messages_check_deps_ClassifyObject
[  0%] Built target diagnostic_msgs_generate_messages_lisp
[  0%] Built target dynamic_reconfigure_generate_messages_eus
[  0%] Built target tf2_msgs_generate_messages_eus
[  0%] Built target tf2_msgs_generate_messages_nodejs
[  0%] Built target tf2_msgs_generate_messages_lisp
[  0%] Built target dynamic_reconfigure_generate_messages_cpp
[  0%] Built target tf2_msgs_generate_messages_py
[  0%] Built target rosgraph_msgs_generate_messages_cpp
[  0%] Built target actionlib_generate_messages_eus
[  0%] Built target actionlib_generate_messages_py
[  0%] Built target tf2_msgs_generate_messages_cpp
[  0%] Built target actionlib_generate_messages_nodejs
[  0%] Built target bond_generate_messages_eus
[  0%] Built target nav_msgs_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target tf_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target roscpp_generate_messages_py
[  0%] Built target bond_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_nodejs
[  0%] Built target rosgraph_msgs_generate_messages_nodejs
[  0%] Built target rosgraph_msgs_generate_messages_py
[  0%] Built target tf_generate_messages_py
[  0%] Built target roscpp_generate_messages_nodejs
[  0%] Built target nav_msgs_generate_messages_py
[  0%] Built target nav_msgs_generate_messages_eus
[  0%] Built target actionlib_msgs_generate_messages_lisp
[  0%] Built target diagnostic_msgs_generate_messages_cpp
[  0%] Built target _realsense2_camera_generate_messages_check_deps_Extrinsics
[  0%] Built target roscpp_generate_messages_eus
[  0%] Built target nav_msgs_generate_messages_nodejs
[  0%] Built target _realsense2_camera_generate_messages_check_deps_IMUInfo
[  0%] Built target actionlib_msgs_generate_messages_cpp
[... 76 lines cut ...]
/usr/include/python2.7/numpy/__multiarray_api.h:1537:144: note: in definition of macro ‘import_array’
 #define import_array() {if (_import_array() < 0) {PyErr_Print(); PyErr_SetString(PyExc_ImportError, "numpy.core.multiarray failed to import"); return NUMPY_IMPORT_ARRAY_RETVAL; } }
                                                                                                                                                ^~~~~~
In file included from /usr/include/python2.7/numpy/ndarrayobject.h:27:0,
                 from /home/ysxt001/openvino_ws/src/vision_opencv/cv_bridge/src/module.hpp:26,
                 from /home/ysxt001/openvino_ws/src/vision_opencv/cv_bridge/src/module.cpp:35:
/home/ysxt001/openvino_ws/src/vision_opencv/cv_bridge/src/module.hpp: In function ‘void* do_numpy_import()’:
/usr/include/python2.7/numpy/__multiarray_api.h:1537:144: error: return-statement with no value, in function returning ‘void*’ [-fpermissive]
 tring(PyExc_ImportError, "numpy.core.multiarray failed to import"); return NUMPY_IMPORT_ARRAY_RETVAL; } }
                                                                     ^
/usr/include/python2.7/numpy/__multiarray_api.h:1537:144: note: in definition of macro ‘import_array’
 tring(PyExc_ImportError, "numpy.core.multiarray failed to import"); return NUMPY_IMPORT_ARRAY_RETVAL; } }
                                                                     ^~~~~~
vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/build.make:86: recipe for target 'vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/module_opencv4.cpp.o' failed
make[2]: *** [vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/module_opencv4.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/build.make:62: recipe for target 'vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/module.cpp.o' failed
make[2]: *** [vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/module.cpp.o] Error 1
CMakeFiles/Makefile2:1293: recipe for target 'vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/all' failed
make[1]: *** [vision_opencv/cv_bridge/src/CMakeFiles/cv_bridge_boost.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

Any idea why I get this error?

LewisLiuPub, 2020-05-21:
hi，currently, vision_opencv/cv_bridge doesn't support openCV 4.x. So the best option for you is to change openCV to 3.x in your build environment.
as an alternative, I have also filed new patch for cv_bridge project to support openCV4.x, you can also try that version, see more from cv_bridge github repo.

HermanLiu1, 2020-07-23:
Thanks a lot! When I change the version of vision_opencv to the branch of melodic(My Ros version is melodic), Everything is ok.

LewisLiuPub, 2020-07-27:
Glad to know that, so close this defect.

closed 2020-07-27 by LewisLiuPub