intel/ros_openvino_toolkit #83: New development features in Y2021

LewisLiuPub, 2020-11-24:
I am re-assigning some efforts from Intel and recruiting from open communities to quickly follow ROS and OpenVINO's latest releases. First I would like to gather requirements from the developers, real users or field experts, then filter them in a feasible development priorities and schedules.

Here are some thoughts about migration/development activities, wish anyone interested can share yours:

- Porting to OpenVINO Toolkit 2021.1
-  Porting to ROS Noetic
- Docker image support

Please feel free to share your thoughts, what features do you want this project have?

Thanks.

WaldoPepper, 2020-12-04:
Support of ROS standard `vision_msgs` would be great.

silasalves, 2020-12-14:
Thanks for continuing the development of ros_openvino_toolkit! We are using it on our robot [Aether](https://3srp.com/) to improve human-robot interaction, so this is good news for us.

A few thoughts on the proposed suggestions:

* **Porting to OpenVINO Toolkit 2021.1** is very important, as some of the new models (e.g. [YOLOv4](https://github.com/TNTWEN/OpenVINO-YOLOV4/)) require more recent versions of OpenVINO Toolkit.

* **Porting to ROS Noetic** is welcome, but it would be good to have backwards compatibility to Kinetic and Melodic. I know that Kinetic is old, but most companies are using robot bases that only supports Kinetic.

* **Docker image support** is welcome. We are currently testing the [openvino/ubuntu18_dev](https://hub.docker.com/r/openvino/ubuntu18_dev) docker into our workflow. Right now we are not sure how to integrate that to our ROS development since it is completely isolated from the rest of the system, but the ease of use sure is appealing.

I talked to my team and we have a few suggestions as well:

* **Make it easier to integrate new models.** We currently added the [human-pose-estimation-0001 model](https://docs.openvinotoolkit.org/2019_R3.1/_models_intel_human_pose_estimation_0001_description_human_pose_estimation_0001.html) to [our repository](https://github.com/3SpheresRoboticsProject/ros_openvino_toolkit/tree/add_skel_track). Although it was not impossible, it was time consuming. A solution could be to create a script that can automatically generate the stub code for the new network, so that the user just have to change those files. Another option would be using [ROS pluginlib](http://wiki.ros.org/pluginlib), but that may overcomplicate the design of ros_openvino_toolkit.

* **Improve the pipeline design.** The current pipeline implementation makes ros_openvino_toolkit very flexible. It’s very useful being able to adapt the package’s functionality by changing a configuration YML file. However, the current implementation doesn’t seem to allow users to create multiple topics or to have both topics and services running in parallel. We have an use-case where we need to run the person re-identification on demand, so we need a pipeline design that would allow us to have both the continuous (topic) and the on-demand (service) solutions. It would be awesome if services could be defined in the pipeline, rather than on a separate application.

**Minor improvements.** Such as:

* considering copying the `stamp` of the input messages to the output message (for example, [this commit](https://github.com/3SpheresRoboticsProject/ros_openvino_toolkit/commit/a129744ac103772aa9af3742acc6ecda91e686dd)). This eases the use of the [message_filters](http://wiki.ros.org/message_filters) package.
* enabling remapping of input names for those not using Intel RealSense.

LewisLiuPub, 2020-12-15:
> Support of ROS standard `vision_msgs` would be great.

Thank you @WaldoPepper very much for your advice. It's really a good idea to support it.

LewisLiuPub, 2020-12-15:
@silasalves Thank you very much for your good and detailed comments. They are really really valuable for me to make a reasonable feature list and dev plan for the coming year. and also appreciate your contribute you have done and to do against the repo.