intel/ros_openvino_toolkit #103: [ERROR] ‘struct PipelineManager::PipelineData’ has no member named ‘parent_node’

Corsair-cxs, 2021-02-26:
### error: ‘struct PipelineManager::PipelineData’ has no member named ‘parent_node’
#### 

#### Environment(for bug reports)
- [x] Operating System: Ubuntu 18.04
- [x] ROS version: Melodic
- [x] ros_openvino_toolkit version: dev-ov2020.3 (newest)

#### My Related code

pipeline_manager.cpp, 
- [ ] function `std::shared_ptr<Pipeline> PipelineManager::createPipeline(const Params::ParamManager::PipelineRawData& params)` 
```cpp
  PipelineData data;
  data.parent_node = node;
```

pipeline_manager.h:
- [ ] PipelineData's defination
```cpp
  struct PipelineData
  {
    Params::ParamManager::PipelineRawData params;
    std::shared_ptr<Pipeline> pipeline;
    std::vector<std::shared_ptr<ros::NodeHandle>> spin_nodes;
    std::shared_ptr<std::thread> thread;
    PipelineState state;
  };
```

#### Current Result 
```
/home/cxs/my_code/ros_vino_ws/src/ros_openvino_toolkit/dynamic_vino_lib/src/pipeline_manager.cpp:72:8: error: ‘struct PipelineManager::PipelineData’ has no member named ‘parent_node’
   data.parent_node = node;
        ^~~~~~~~~~~
/home/cxs/my_code/ros_vino_ws/src/ros_openvino_toolkit/dynamic_vino_lib/src/pipeline_manager.cpp:72:22: error: ‘node’ was not declared in this scope
   data.parent_node = node;
                      ^~~~
```

#### It seems that `parent_node` was defined in ros2_openvino_toolkit's pipeline_manager.h while not defined in this project.

LewisLiuPub, 2021-03-01:
Thanks for the contribution.

just using ros node handler to replace of ros2 node instance, when pub/sub messages.

Corsair-cxs, 2021-03-01:
> Thanks for the contribution.
> 
> just using ros node handler to replace of ros2 node instance, when pub/sub messages.

Okey, thanks!