intel/ros_openvino_toolkit #150: ROS Noetic support?

raghavauppuluri13, 2021-03-23:
When would there be support for ROS Noetic Ubuntu 20.04?

Corsair-cxs, 2021-03-24:
May be in the next few months.

Corsair-cxs, 2021-03-24:
> When would there be support for ROS Noetic Ubuntu 20.04?

May be in the next few months.

LewisLiuPub, 2021-03-24:
Thanks for your attention, @raghavauppuluri13, and your inputs, @Corsair-cxs 

Currently there are some volunteer students, including @Corsair-cxs, co-working with me to migrate new versions of OpenVINO and support new ROS version.
@raghavauppuluri13 , could you let me know more about your uses of OpenVINO, 
- what features of OpenVINO would you use, 
- what product are you working with OpenVINO engaged? 
- how urgent do you need the Neotic-20.04 features?
- etc. 
Thanks a lot.

shubhamwagh, 2021-08-09:
Hi, any update on support for ROS Noetic Ubuntu 20.04?

LewisLiuPub, 2021-08-20:
@shubhamwagh, we are now still in the development phase. We are paying attention to your request, and will let you know once it is ready for release.

julianx4, 2021-12-22:
Any eta on this?

LewisLiuPub, 2022-01-25:
ready!, see branch [noetic](https://github.com/intel/ros_openvino_toolkit/tree/noetic).

closed 2022-01-25 by LewisLiuPub

BRoboticsSHADO, 2022-07-18:
Hi @LewisLiuPub

For the inferenceEngine, which branch shall i use? so that it will be compatible with noetic branch of ros_openvino_toolkit?

Best,
Samuel

LewisLiuPub, 2022-07-18:
@BRoboticsSHADO , we tested OpenVINO 2021.4LTS version (e.g. **2021.4.2LTS**).
You can download them by "apt install" commands, see the [OpenVINO download page](https://www.intel.com/content/www/us/en/developer/tools/openvino-toolkit/download.html) for more details.
For the case of using Ubuntu and APT installation env, see [this page](https://docs.openvino.ai/nightly/openvino_docs_install_guides_installing_openvino_apt.html)