intel/ros_openvino_toolkit #173: Run with lates librealsense and ros image topic

Lilienthal-GF, 2021-05-11:
Hello everyone,
i am having trouble getting the ros_openvino_toolkit package to run with the dev-ov2020.3 branch. 

The platform is built on Ubuntu 18.04.5 LTS with the 5.4 Kernel. Therefore i would like to run the lates version of librealsense 2.45.0 and the corresponding realsense-ros package 2.3.0. 

I installed the openvino 2020.3 release and confirmed the demo applications on the setup and i can succesfully run catkin_make on the workspace. But if i launch the 

```shell
roslaunch vino_launch pipeline_object_topic.launch
```

the realsense camera gets launched correctly (i can view the image in the rostopic), yet i get the error message 

```shell
[ INFO ] Parsing InputDvice: RealSenseCameraTopic
[ ERROR ] Invalid input device name: RealSenseCameraTopic
[ ERROR ] currently one pipeline only supports ONE input.
```

I traced this back to the pipeline_manager.cpp and discovered that the RealSenseCameraTopic is commented out in the dev-ov2020.3 branch of ros_openvino_toolkit. Is there a way to fix this?

Corsair-cxs, 2021-05-13:
Thank you for your reportation!
I've tried to fix the bug in #174 , could you please test the [new pr](https://github.com/Corsair-cxs/ros_openvino_toolkit/tree/iss174) it in your computer?

Best Wishes

Lilienthal-GF, 2021-05-17:
I can confirm that merging the commit fixes the problem. I changed the pipeline_object_topic.launch to launch the rs_camera-node again. But now i the following error message when i launch the pipeline_object_topic.launch 

```Shell
[ ERROR ] Unexpected CNNNetwork format: it was converted to deprecated format prior plugin's call
[pipeline_with_params-4] process has died [pid 17657, exit code 1
```

I downloaded and converted the mobile-ssd from the open_model_zoo in the openvino install folder, i.e.
```Shell
sudo python3 ~/intel/openvino/deployment_tools/model_optimizer/mo.py --input_model ~/intel/openvino/deployment_tools/open_model_zoo/tools/downloader/public/mobilenet-ssd/mobilenet-ssd.caffemodel --output_dir /opt/openvino_toolkit/models/object_detection/mobilenet-ssd/caffe/output/FP16 --data_type=FP16 --mean_values [127.5,127.5,127.5] --scale_values [127.5]
```

Do i need to use a different tool to download or convert the net?

LewisLiuPub, 2021-08-20:
I think you need to check the version of OpenVINO Toolkit you installed. branch **dev-ov2020.3** is now the latest stable branch, you also need to use the same version (2020.3LTS) of Intel OpenVINO toolkit (including Inference Engine and Model Optimizer).
Please try and let us know your results. @Lilienthal-GF