intel/ros_openvino_toolkit #180: Please help Raspberry users for god sake!

gholibeigian, 2021-08-20:
Dear developers,

I use RPI4-MODBP-8GB for my thesis. I bought Intel Neural Compute Stick 2 for my project.

I know you don't have a lot of time writing instructions and you spend so much time writing documentation.
I'm grateful for that, but there is a problem with your project!

None of your instructions work on Raspberry Pi 4. I'm a student and have tried everything to install OpenCV and OpenVino on my Raspberry Pi. I was able to do it last year, but I can't reinstall it with a new version of Raspbian!

I have a good idea for your company! Most Raspberry Pi users are students (we are not professionals like you)
Can you please install OpenCV and Openvino on a Raspberry Pi image and put them on your website ?! So that we can download and use it!
There aren't many Raspberry Pi variants and your sales will increase! More students want to buy such a device with such ready-to-use images!

Or at least write a correct instructions on your website!

I followed these instructions and many combinations of them:

https://www.intel.com.au/content/www/au/en/support/articles/000057005/boards-and-kits.html

https://software.intel.com/content/www/us/en/develop/articles/arm-sbc-and-ncs2.html

https://software.intel.com/content/www/us/en/develop/articles/intel-neural-compute-stick-2-and-open-source-openvino-toolkit.html

https://www.intel.com/content/www/us/en/support/articles/000057005/boards-and-kits.html

None of them has worked and I cannot buy a new stick from Intel. I am a student and not rich (I think your know what I mean)

Best regards,
Kazem Gholibeigian

LewisLiuPub, 2021-08-24:
Dear Kazem , @gholibeigian 
Thank you very much for attention and requests. 
I am transferring your demands to more OpenVINO developers and deciders. But in parallel, I want to get more info from you:
- Do you purely use Intel OpenVINO toolkit or do you use both OpenVINO and ROS system?
-  What AI features will you use OpenVINO on your  Raspberry Pi platform? 

Your answer will help us evaluate the feasibility of computing and development.
Thanks

Best Regards
Weizhi Liu

gholibeigian, 2021-08-24:
> 
> 
> Dear Kazem , @gholibeigian
> Thank you very much for attention and requests.
> I am transferring your demands to more OpenVINO developers and deciders. But in parallel, I want to get more info from you:
> 
>     * Do you purely use Intel OpenVINO toolkit or do you use both OpenVINO and ROS system?
> 
>     * What AI features will you use on your  Raspberry Pi platform?
> 
> 
> Your answer will help us evaluate the feasibility of computing and development.
> Thanks
> 
> Best Regards
> Weizhi Liu

Dear Lewis Liu,

I used Opencv's built-in hair cascade on Raspberry Pi 4 to see if anyone was in front of my camera. When someone is in front of the camera, my thermal camera will start taking the person's thermal images! I had a problem! The Raspberry Pi CPU wasn't that powerful. So my raspberry has crashed! That's why I bought the Neural Compute Stick 2. NC2 should do the calculation for face detection. So the Raspberry can dedicate its resources to more important tasks in my project.

My code in python3 is something like this:

`import datetime
import subprocess
import time
import cv2 as cv
# Load the model.
net = cv.dnn.readNet('face-detection-adas-0001.xml',
                     'face-detection-adas-0001.bin')
# Specify target device
net.setPreferableTarget(cv.dnn.DNN_TARGET_MYRIAD)
# Read an image - used an extracted image the open source video: Camera(0)
cap = cv.VideoCapture(0)
# Check to see if the camera works
if not cap.isOpened():
    print("Cannot open camera")
    exit()
while True:

    ret, frame = cap.read()
    if not ret:
        print("Can't receive frame (stream end?). Exiting ...")
        break

    # Prepare input blob and perform an inference.
    blob = cv.dnn.blobFromImage(frame, size=(672, 384), ddepth=cv.CV_8U)
    net.setInput(blob)
    out = net.forward()
    # Draw detected faces on the frame.
    for detection in out.reshape(-1, 7):
        confidence = float(detection[2])
        xmin = int(detection[3] * frame.shape[1])
        ymin = int(detection[4] * frame.shape[0])
        xmax = int(detection[5] * frame.shape[1])
        ymax = int(detection[6] * frame.shape[0])
        if confidence > 0.5:
            #cv.rectangle(frame, (xmin, ymin), (xmax, ymax), color=(0, 255, 0))
            # I am going to use subprocess to capture thermal images here:
`

However it is not possible to use your Opencv-dnns in the code! It means Python3 on Raspberry pi 4( RPI4-MODBP-8GB) cannot load your face detection model on this line:

# Load the model.
net = cv.dnn.readNet('face-detection-adas-0001.xml',
                     'face-detection-adas-0001.bin')

I just need to do the programming in python3 on Raspbian and your Neural Compute Stick 2 does not work on the new version of Raspbian!
I have already followed all your instructions since one month ego, but I could not manage it.

Best regards,
Kazem

LewisLiuPub, 2021-09-01:
sorry for late reply. @gholibeigian 
Judging from the code you shared, you are using cv.dnn framework, not OpenVINO framework. and you are not using ROS/ROS2 system. cv.dnn is not covered by my team.
and I tried to find a contact window of CV.dnn for you, but failed. So I recommend you get some response directly from [CV.dnn forum](https://forum.opencv.org/).

Thanks.
Weizhi