intel/ros_openvino_toolkit #182: 2021.4 and IE version incompatibility issue

sreshu, 2021-08-24:
While doing catkin_make, we get this issue-

CMake Warning at ros_openvino_toolkit/sample/CMakeLists.txt:20 (find_package):
  Could not find a configuration file for package "InferenceEngine" that is
  compatible with requested version "1.1".

  The following configuration files were considered but not accepted:

    /opt/intel/openvino_2021/deployment_tools/inference_engine/share/InferenceEngineConfig.cmake, version: 2021.4.0

CMake Error at ros_openvino_toolkit/sample/CMakeLists.txt:22 (message):

-- Configuring incomplete, errors occurred!
See also "/home/reshu/Desktop/openvino_exp/vino_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/reshu/Desktop/openvino_exp/vino_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

LewisLiuPub, 2021-09-01:
@sreshu Please try with new branch: **dev-ov2021.4**. Happy to hear your comments after your attempts..

LewisLiuPub, 2022-01-25:
branches melodic and noetic are verified.

closed 2022-01-25 by LewisLiuPub