intel/ros_openvino_toolkit #199: cmake error

kethan637, 2024-05-16:
kethan@kethan:~/ros2_ws$ colcon build --symlink-install
Starting >>> object_analytics_msgs
Starting >>> ros2_workspace
--- stderr: object_analytics_msgs                                                             
CMake Error at CMakeLists.txt:32 (find_package):
  By not providing "Findobject_msgs.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "object_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "object_msgs" with
  any of the following names:

    object_msgsConfig.cmake
    object_msgs-config.cmake

  Add the installation prefix of "object_msgs" to CMAKE_PREFIX_PATH or set
  "object_msgs_DIR" to a directory containing one of the above files.  If
  "object_msgs" provides a separate development package or SDK, be sure it
  has been installed.

kethan637, 2024-05-16:
is this work with ros2 humble or not if yes can you send me steps for that

LewisLiuPub, 2024-05-17:
object-msgs is a dependent package, you need to clone its source code before building.

ROS: https://github.com/intel/ros_object_msgs
ROS2: https://github.com/intel/ros2_object_msgs

Liusdaooo, 2024-05-20:
我有一段CMakeTestCXXCompiler.cmake代码 在把YOLOV8部署到Android studio手机上时，报CMake Error at D:/AndroidSDK/cmake/3.10.2.4988404/share/cmake-3.10/Modules/CMakeTestCXXCompiler.cmake:45 (message):
The C++ compiler 怎么解决？ 请教大佬们，求你们了 这个项目对我比较重要！   关键代码如下：message(FATAL_ERROR "The C++ compiler\n  \"${CMAKE_CXX_COMPILER}\"\n"
    "is not able to compile a simple test program.\nIt fails "
    "with the following output:\n  ${_output}\n\n"
    "CMake will not be able to correctly generate this project.")
  message(FATAL_ERROR "The C compiler\n \"${CMAKE_C_COMPILER}\"\n"
    "is not able to compile a simple test program.\nIt fails "
    "with the following output:\n  ${_output}\n\n"
    "CMake will not be able to correctly generate this project.")